balena-io/etcher #2397: Empty grey window when starting in Ubuntu 17.10

Apachez-, 2018-06-18:
Using Ubuntu 17.10, installed Etcher 1.4.4 according to instructions at https://github.com/resin-io/etcher (Debian and Ubuntu based Package Repository (GNU/Linux x86/x64)) and verified that it is installed:

user@nuc:~$ dpkg -l | grep -i etcher
ii  etcher-electron                            1.4.4                                       amd64        Etcher is a powerful OS image flasher built with web technologies to ensure flashing an SDCard or USB drive is a pleasant and safe experience. It protects you from accidentally writing to your hard-drives, ensures every byte of data was written correctly and much more.

When starting Etcher I receive just an empty grey window (se attached picture):

[image]

When starting developer console (ctrl+shit+I) I get this output which might give you a hint on what the error might be?

Uncaught Error: /tmp/user/1000/.org.chromium.Chromium.pUVpcf: failed to map segment from shared object
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:187:18)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at /opt/Etcher/resources/app.asar/node_modules/lzma-native/index.js:11:14
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/node_modules/lzma-native/index.js:623:3)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/node_modules/lzma-native/index.js:625:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/sdk/image-stream/handlers.js:24:36)
    at Object.<anonymous> (/opt/Etcher/resources/app.asar/lib/sdk/image-stream/handlers.js:255:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)

Apachez-, 2018-06-18:
As a workaround the CLI version seems to be working on the very same machine/installation:

etcher-cli-1.4.4-linux-x64.tar.gz

BrettRyland, 2018-06-20:
I get the same on Debian buster/sid (Linux debian 4.16.0-2-amd64 #1 SMP Debian 4.16.12-1 (2018-05-27) x86_64 GNU/Linux).
Incidentally, running `ldd` on etcher shows a couple of missing dynamic library dependencies, though installing them seems to have no effect on this issue.
```
brett@debian:~$ ldd $(which etcher-electron)
        linux-vdso.so.1 (0x00007ffdfc996000)
        libnode.so => not found
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe3e5edf000)
        libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007fe3e589d000)
        libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007fe3e55e8000)
        libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007fe3e53db000)
        libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007fe3e51b5000)
        libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007fe3e4e98000)
        libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007fe3e4c74000)
        libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007fe3e48d6000)
        libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007fe3e4691000)
        libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fe3e443d000)
        libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fe3e4188000)
        libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007fe3e3f45000)
        libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fe3e3cf4000)
        libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fe3e3af2000)
        libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fe3e38ca000)
        libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007fe3e36ba000)
        libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007fe3e34b0000)
        libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fe3e32ad000)
        libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007fe3e30a2000)
        libXcomposite.so.1 => /usr/lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007fe3e2e9f000)
        libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fe3e2c8d000)
        libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fe3e2a87000)
        libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fe3e287d000)
        libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fe3e253f000)
        libXtst.so.6 => /usr/lib/x86_64-linux-gnu/libXtst.so.6 (0x00007fe3e2339000)
        libXss.so.1 => /usr/lib/x86_64-linux-gnu/libXss.so.1 (0x00007fe3e2136000)
        libgconf-2.so.4 => /usr/lib/x86_64-linux-gnu/libgconf-2.so.4 (0x00007fe3e1f05000)
        libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007fe3e1d01000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe3e1af9000)
        libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fe3e17e3000)
        libnss3.so => /usr/lib/x86_64-linux-gnu/libnss3.so (0x00007fe3e149a000)
        libnssutil3.so => /usr/lib/x86_64-linux-gnu/libnssutil3.so (0x00007fe3e126a000)
        libsmime3.so => /usr/lib/x86_64-linux-gnu/libsmime3.so (0x00007fe3e103d000)
        libnspr4.so => /usr/lib/x86_64-linux-gnu/libnspr4.so (0x00007fe3e0dfe000)
        libffmpeg.so => not found
        libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007fe3e0b00000)
        libcups.so.2 => /usr/lib/x86_64-linux-gnu/libcups.so.2 (0x00007fe3e0871000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe3e066d000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fe3e043b000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe3e00ba000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe3dfd27000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe3dfb0f000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe3df755000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fe3e60fd000)
        libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007fe3df540000)
        libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007fe3df33d000)
        libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007fe3df097000)
        libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fe3dee64000)
        libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007fe3dec61000)
        libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007fe3dea53000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fe3de835000)
        libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007fe3de60d000)
        libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007fe3de3f6000)
        libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007fe3de19d000)
        libthai.so.0 => /usr/lib/x86_64-linux-gnu/libthai.so.0 (0x00007fe3ddf93000)
        libfribidi.so.0 => /usr/lib/x86_64-linux-gnu/libfribidi.so.0 (0x00007fe3ddd7c000)
        libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fe3ddb73000)
        libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fe3dd96c000)
        libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fe3dd6e5000)
[... cut ...]
        libtasn1.so.6 => /usr/lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007fe3d97e3000)
        libnettle.so.6 => /usr/lib/x86_64-linux-gnu/libnettle.so.6 (0x00007fe3d95ac000)
        libhogweed.so.4 => /usr/lib/x86_64-linux-gnu/libhogweed.so.4 (0x00007fe3d9376000)
        libgmp.so.10 => /usr/lib/x86_64-linux-gnu/libgmp.so.10 (0x00007fe3d90f2000)
        libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fe3d8ec5000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fe3d8ca5000)
```

hawkeyese, 2019-01-10:
I have the same symptom on Ubuntu 18.10 running on a ThinkPad t430s (integrated Intel gfxcard).
Has anybody found a way around this?

hawkeyese, 2019-01-11:
Ah, that's probably my problem. I have /tmp mounted noexec.

lurch, 2019-01-12:
@thundron @zvin I wonder if it's possible to detect this condition and display an appropriate error message, rather than just showing am empty grey window? :man_shrugging:

lurch, 2019-01-12:
@hawkeyese Is that the out-of-the-box behaviour in Ubuntu 18.10, or did you perform some kind of custom configuration step?

hawkeyese, 2019-01-12:
It is my own customization. Feels somewhat safer.

aasril, 2019-06-20:
I had the same problem with a newly installed Etcher on Linux Mint 18.3 (Etcher 1.5.50). 

I'm running out of space on my `/home` partition and I decided to do a fresh install on a new SSD. After installing Etcher, it only showed me a blank window (but with settings button and others on that line visible). Apparently, either installation of Etcher or its first launch, it ate the remaining of my `/home`. 

After moving some files to an external drive and logged out, Etcher ran and wrote the image successfully. I'm not sure whether it caused by insufficient space or I simply need to log out.

closed 2021-08-17 by thundron

Apachez-, 2021-08-23:
Even if the same machine is currently at Ubuntu 21.04 I can confirm that remounting /tmp with noexec removed resolved this issue.

fstab with this didnt work:

```
tmpfs  /tmp  tmpfs  defaults,nodev,nosuid,noexec  0  0
tmpfs  /var/tmp  tmpfs  defaults,nodev,nosuid,noexec  0  0

```
But with this it does work:

```
tmpfs  /tmp  tmpfs  defaults,nodev,nosuid  0  0
tmpfs  /var/tmp  tmpfs  defaults,nodev,nosuid  0  0

```