balena-io/etcher #2398: Fails to detect USB drive from Docker container

breandan, 2018-06-19:
- **Etcher version:** 1.4.4 
- **Operating system and architecture:** Debian GNU/Linux 9 (running on Docker version 18.03.1-ce, build 9ee9f40)
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** No.

`etcher-cli` fails to detect a USB drive passed through to a Docker container, via `docker run -it --privileged -v /dev/bus/usb:/dev/bus/usb duckietown/image-builder`. Displays the following message:

```
x No available drives were detected, plug one in!
```

This is the output of `lsblk`:

```
root@7bae958062e3:/go# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0         7:0    0 364.8M  1 loop 
loop1         7:1    0 364.1M  1 loop 
loop2         7:2    0 365.2M  1 loop 
loop3         7:3    0  86.6M  1 loop 
loop4         7:4    0 591.1M  1 loop 
loop5         7:5    0 143.9M  1 loop 
loop6         7:6    0 590.7M  1 loop 
loop7         7:7    0 565.6M  1 loop 
loop8         7:8    0  93.5M  1 loop 
sda           8:0    1  59.5G  0 disk 
|-sda1        8:1    1    64M  0 part 
`-sda2        8:2    1  59.4G  0 part 
nvme0n1     259:0    0   477G  0 disk 
|-nvme0n1p1 259:1    0   499M  0 part 
|-nvme0n1p2 259:2        100M  0 part 
|-nvme0n1p3 259:3    0    16M  0 part 
|-nvme0n1p4 259:4       48.2G  0 part 
|-nvme0n1p5 259:5    0 412.3G  0 part /etc/hosts
`-nvme0n1p6 259:6    0  15.9G  0 part 
```

Expected result is `etcher-cli` should have detected `sda` as a valid USB device.

lurch, 2018-06-19:
It might be worth opening an issue for this on https://github.com/resin-io-modules/drivelist (which is the module Etcher uses to list drives)

breandan, 2018-06-19:
While #1420 seems like a viable workaround, I don't think this is a duplicate, but some issue with USB drive detection that causes it to fail inside Docker environments.