balena-io/etcher #2399: Error message when trying to add zip file

Murphyv310, 2018-06-19:
- **Etcher version:** 1.43x86_64
- **Operating system and architecture:** Linux Mint 18.3 Cinnamon 64 bit
- **Image flashed:** Noobs V2_8_1.zip
- **Do you see any meaningful error information in the DevTools?** 

I get this error, it has worked OK in the past but new downloads seem to be more problematical, same issue on another PC using the Mint 18.3 but 32 bit 

/home/trevor/Pictures/Screenshot from 2018-06-19 13-21-06.png

jhermsmeier, 2018-06-19:
NOOBS is not a disk image (and hence Etcher can't flash it), but an archive of a boot partition; please see #1756 for some background, and refer to the [Raspberry Pi Foundation's documentation](https://www.raspberrypi.org/documentation/installation/noobs.md).

Duplicate of https://github.com/resin-io/etcher/issues/1162

jhermsmeier, 2018-06-19:
Duplicate of https://github.com/resin-io/etcher/issues/1162

closed 2018-06-19 by jhermsmeier

Murphyv310, 2018-06-19:
I see,  thanks for the explanation. 
Back to the drawing board.

jhermsmeier, 2018-06-19:
@Murphyv310 you can flash the other Raspbian images with Etcher though, unless you need / want to use NOOBS

Murphyv310, 2018-06-19:
Thank you.
I've a 3b+ and I've found out that Ubuntu Mate doesn't load, my previous 3B was OK. To be honest all I want to run is GQRX  and other SDR software but the 3B+ doesn't play ball,  feeling like giving up on the Pi now.

lurch, 2018-06-20:
> I've a 3b+ and I've found out that Ubuntu Mate doesn't load, my previous 3B was OK.

I've not used Ubuntu Mate on the RPi myself, but the 3B+ needs newer firmware blobs than the 3B. I did a quick search and there seems to be multiple topics about the 3B+ at https://ubuntu-mate.community/c/support/raspberry-pi-2