balena-io/etcher #2404: Crashing on macOS Mojave Beta 2

IAmOrion, 2018-06-21:
- **Etcher version:** 

1.4.4

- **Operating system and architecture:** 

macOS Mojave 10.14 Beta 2

- **Image flashed:** 

motioneyeos-raspberrypi-20180602.img

- **Do you see any meaningful error information in the DevTools?** 

DevTools closes with etcher when it crashes, so can only provide the os report

macOC Report:

`Process:               Etcher [23059]
Path:                  /Applications/Etcher.app/Contents/MacOS/Etcher
Identifier:            io.resin.etcher
Version:               1.4.4 (1.4.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Etcher [23059]
User ID:               501

Date/Time:             2018-06-22 00:25:51.379 +0100
OS Version:            Mac OS X 10.14 (18A314h)
Report Version:        12
Anonymous UUID:        32698C62-1220-E1A2-A856-2DFDA089401C

Sleep/Wake UUID:       A5DACF2E-3D30-43CC-8B82-7EDB08E18988

Time Awake Since Boot: 150000 seconds
Time Since Wake:       1600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [23059]

Application Specific Information:
Crashing on exception: -[_NSImageViewSimpleImageView setDoubleValue:]: unrecognized selector sent to instance 0x600001908460

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff503a9600 __exceptionPreprocess + 197
1   libobjc.A.dylib                     0x00007fff7c56b933 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff50432ba1 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
3   CoreFoundation                      0x00007fff50336c61 ___forwarding___ + 1489
4   CoreFoundation                      0x00007fff50336608 _CF_forwarding_prep_0 + 120
5   Electron Framework                  0x000000010f4605ae _ZN4atom15NativeWindowMac14SetProgressBarEdNS_12NativeWindow13ProgressStateE + 606
6   Electron Framework                  0x000000010f432851 _ZN4atom3api6Window14SetProgressBarEdPN4mate9ArgumentsE + 433
7   Electron Framework                  0x000000010f4364f5 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 149
8   Electron Framework                  0x000000010f4364cc _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 108
9   Electron Framework                  0x000000010f4364a2 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 66
10  Electron Framework                  0x000000010f43647d _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 29
11  Electron Framework                  0x000000010f4363fd _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 589
12  Electron Framework                  0x000000010f436351 _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 417
13  Electron Framework                  0x000000010f43622c _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 124
14  libnode.dylib                       0x0000000113ca9f21 _ZN2v88internal6StrDupEPKc + 609
15  libnode.dylib                       0x0000000113d5ece8 _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 4136
16  libnode.dylib                       0x0000000113d5e29b _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 1499
[... 925 lines cut ...]
===========                     =======  ======= 
TOTAL                              2.1G     1099 
TOTAL, minus reserved VM space     1.8G     1099 

Model: MacBookPro10,1, BootROM MBP101.00F6.B00, 4 processors, Intel Core i7, 2.6 GHz, 8 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533642465238432D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533642465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.190.30 AirPortDriverBrcm4360-1301)
Bluetooth: Version 6.0.8d23, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM512E, 500.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
`

IAmOrion, 2018-06-22:
Do'h totally forgot to write that (it was super late lol) ... The crash happens the instant you press "Flash".

I can pick a file no problem, I can pick the SD Card (or it's auto picked if only 1 flash media detected), but when I click "Flash" it crashes and I get the report as mentioned

rv10guy, 2018-06-24:
It will also crash when simply selecting Etcher...About

IAmOrion, 2018-06-24:
> It will also crash when simply selecting Etcher...About

Not for me it doesn't

[image]

SharanSrinivasan, 2018-06-26:
I have this problem too!

concept10, 2018-06-29:
Reporting same issue here.  macOS Mohave 10.14 Beta (18A314h), Etcher 1.4.4.

vectorhacker, 2018-06-30:
Same issue here. Here's my log:

```
Process:               Etcher [9849]
Path:                  /Applications/Etcher.app/Contents/MacOS/Etcher
Identifier:            io.resin.etcher
Version:               1.4.4 (1.4.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Etcher [9849]
User ID:               501

Date/Time:             2018-06-29 23:52:09.379 -0400
OS Version:            Mac OS X 10.14 (18A314h)
Report Version:        12
Anonymous UUID:        3BF90A9F-4C5B-6627-8888-D2834AAF8EA3

Sleep/Wake UUID:       41A1356E-51B0-4466-ABFC-97317FE785B9

Time Awake Since Boot: 34000 seconds
Time Since Wake:       15000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [9849]

Application Specific Information:
Crashing on exception: -[_NSImageViewSimpleImageView setDoubleValue:]: unrecognized selector sent to instance 0x60000154a260

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff411f1484 __exceptionPreprocess + 197
1   libobjc.A.dylib                     0x00007fff6d31ed08 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff4127a15f -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
3   CoreFoundation                      0x00007fff4117efd2 ___forwarding___ + 1506
4   CoreFoundation                      0x00007fff4117e968 _CF_forwarding_prep_0 + 120
5   Electron Framework                  0x000000010209c5ae _ZN4atom15NativeWindowMac14SetProgressBarEdNS_12NativeWindow13ProgressStateE + 606
6   Electron Framework                  0x000000010206e851 _ZN4atom3api6Window14SetProgressBarEdPN4mate9ArgumentsE + 433
7   Electron Framework                  0x00000001020724f5 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 149
8   Electron Framework                  0x00000001020724cc _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 108
9   Electron Framework                  0x00000001020724a2 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 66
10  Electron Framework                  0x000000010207247d _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 29
11  Electron Framework                  0x00000001020723fd _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 589
12  Electron Framework                  0x0000000102072351 _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 417
13  Electron Framework                  0x000000010207222c _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 124
14  libnode.dylib                       0x00000001068f6f21 _ZN2v88internal6StrDupEPKc + 609
15  libnode.dylib                       0x00000001069abce8 _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 4136
16  libnode.dylib                       0x00000001069ab29b _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 1499
17  ???                                 0x000016a59460437d 0x0 + 24900414751613

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   com.apple.AppKit              	0x00007fff3e9fbd88 -[NSApplication _crashOnException:] + 109
1   com.apple.AppKit              	0x00007fff3e9fbc63 -[NSApplication reportException:] + 959
2   com.apple.AppKit              	0x00007fff3ead84df uncaughtErrorProc + 163
3   com.apple.CoreFoundation      	0x00007fff41273855 __handleUncaughtException + 775
4   libobjc.A.dylib               	0x00007fff6d321285 _objc_terminate() + 91
5   libc++abi.dylib               	0x00007fff6bcbc782 std::__terminate(void (*)()) + 8
6   libc++abi.dylib               	0x00007fff6bcbc24f __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
7   libc++abi.dylib               	0x00007fff6bcbc212 __cxa_throw + 113
8   libobjc.A.dylib               	0x00007fff6d31ee42 objc_exception_throw + 362
9   com.apple.CoreFoundation      	0x00007fff4127a15f -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
10  com.apple.CoreFoundation      	0x00007fff4117efd2 ___forwarding___ + 1506
11  com.apple.CoreFoundation      	0x00007fff4117e968 _CF_forwarding_prep_0 + 120
[... 843 lines cut ...]
IOKit                             7940K        2 
Image IO                            64K        2 
Kernel Alloc Once                    8K        2 
MALLOC                           241.8M       41 
MALLOC guard page                   32K        8 
MALLOC_NANO (reserved)           384.0M        2         reserved VM address space (unallocated)
Memory Tag 242                      12K        2 
Memory Tag 251                      24K        2 
Memory Tag 255                   525.0M       58 
[... cut ...]
__DATA                            44.4M      355 
__FONT_DATA                          4K        2 
__LINKEDIT                       218.5M       15 
__TEXT                           330.4M      352 
__UNICODE                          560K        2 
mapped file                       82.5M       87 
shared memory                      772K       15 
===========                     =======  ======= 
TOTAL                              2.0G     1107 
TOTAL, minus reserved VM space     1.7G     1107 

```

berombau, 2018-06-30:
Same issue with crashing Etcher GUI on Mojave when selecting Flash. The [CLI version](https://etcher.io/cli/) however did work.

hexalellogram, 2018-07-07:
Reporting the same issue here, with Etcher 1.4.4, now on macOS Mojave Developer Beta 3 (build 18A326g). Etcher crashes when I click flash.

jakubzitny, 2018-07-13:
This is caused by [BrowserWindow#setProgressBar()](https://electronjs.org/docs/api/browser-window#winsetprogressbarprogress-options), details in https://github.com/electron/electron/issues/13564#issuecomment-404824409.

p2made, 2018-07-13:
I also experience Etcher crashing when I click Flash.
* macOS Mojave Public Beta 3.
* Etcher 1.4.4

jhermsmeier, 2018-07-13:
Interesting, thanks @jakubzitny, and thanks all for bringing this up! I've not gotten my hands on the beta yet, and thus am unable to reproduce at this point – we'll be looking into this though.

luisleonardo, 2018-07-20:
Same here. Crashes when I click flash.

RyanMorash, 2018-07-22:
Same issue on Beta 4:
```
Process:               Etcher [25861]
Path:                  /Applications/Etcher.app/Contents/MacOS/Etcher
Identifier:            io.resin.etcher
Version:               1.4.4 (1.4.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Etcher [25861]
User ID:               501

Date/Time:             2018-07-22 17:01:04.375 -0400
OS Version:            Mac OS X 10.14 (18A336e)
Report Version:        12
Bridge OS Version:     3.0 (14Y664)
Anonymous UUID:        4F291AFA-6690-6952-E0E3-018409F433A4

Sleep/Wake UUID:       CEC5E48B-45C4-4D9C-82FD-DEBFA0F642F6

Time Awake Since Boot: 160000 seconds
Time Since Wake:       100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [25861]

Application Specific Information:
Crashing on exception: -[_NSImageViewSimpleImageView setDoubleValue:]: unrecognized selector sent to instance 0x60000039b700

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff36a9893b __exceptionPreprocess + 256
1   libobjc.A.dylib                     0x00007fff62a73a18 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff36b20fa8 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
3   CoreFoundation                      0x00007fff36a279b2 ___forwarding___ + 1490
4   CoreFoundation                      0x00007fff36a27358 _CF_forwarding_prep_0 + 120
5   Electron Framework                  0x000000010269b5ae _ZN4atom15NativeWindowMac14SetProgressBarEdNS_12NativeWindow13ProgressStateE + 606
6   Electron Framework                  0x000000010266d851 _ZN4atom3api6Window14SetProgressBarEdPN4mate9ArgumentsE + 433
7   Electron Framework                  0x00000001026714f5 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 149
8   Electron Framework                  0x00000001026714cc _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 108
9   Electron Framework                  0x00000001026714a2 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 66
10  Electron Framework                  0x000000010267147d _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 29
11  Electron Framework                  0x00000001026713fd _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 589
12  Electron Framework                  0x0000000102671351 _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 417
13  Electron Framework                  0x000000010267122c _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 124
14  libnode.dylib                       0x0000000106eeff21 _ZN2v88internal6StrDupEPKc + 609
15  libnode.dylib                       0x0000000106fa4ce8 _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 4136
16  libnode.dylib                       0x0000000106fa429b _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 1499
17  ???                                 0x00000411e178437d 0x0 + 4474843710333

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   com.apple.AppKit              	0x00007fff342728fc -[NSApplication _crashOnException:] + 109
1   com.apple.AppKit              	0x00007fff342727df -[NSApplication reportException:] + 939
2   com.apple.AppKit              	0x00007fff3434e1b2 uncaughtErrorProc + 157
3   com.apple.CoreFoundation      	0x00007fff36b1a769 __handleUncaughtException + 775
4   libobjc.A.dylib               	0x00007fff62a75f95 _objc_terminate() + 91
5   libc++abi.dylib               	0x00007fff612797d2 std::__terminate(void (*)()) + 8
6   libc++abi.dylib               	0x00007fff612792a0 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
7   libc++abi.dylib               	0x00007fff61279263 __cxa_throw + 113
8   libobjc.A.dylib               	0x00007fff62a73b52 objc_exception_throw + 362
9   com.apple.CoreFoundation      	0x00007fff36b20fa8 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
10  com.apple.CoreFoundation      	0x00007fff36a279b2 ___forwarding___ + 1490
11  com.apple.CoreFoundation      	0x00007fff36a27358 _CF_forwarding_prep_0 + 120
[... 940 lines cut ...]
__FONT_DATA                          4K        2 
__LINKEDIT                       219.8M       15 
__TEXT                           374.3M      376 
__UNICODE                          560K        2 
mapped file                       93.4M       84 
shared memory                     2964K       17 
===========                     =======  ======= 
TOTAL                              3.2G     1209 
TOTAL, minus reserved VM space     2.8G     1209 

[... cut ...]
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x173), Broadcom BCM43xx 1.0 (7.77.60.1 AirPortDriverBrcmNIC-1304.1)
Bluetooth: Version 6.0.8d27, 3 services, 27 devices, 1 incoming serial ports
Network Service: Belkin USB-C LAN, Ethernet, en8
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: USB 2.0 FD
USB Device: iBridge
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1
```

atulkakrana, 2018-07-24:
same issue here

Dygear, 2018-07-26:
```dygear@MarkBookPro:/Applications/Etcher.app/Contents/MacOS$ sudo ./Etcher
2018-07-26 17:29:36.265 Etcher[13202:2853911] *** WARNING: Textured window <AtomNSWindow: 0x7fcaee4f7f30> is getting an implicitly transparent titlebar. This will break when linking against newer SDKs. Use NSWindow's -titlebarAppearsTransparent=YES instead.
objc[13202]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff9066f8b8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x11cd0edc0). One of the two will be used. Which one is undefined.
objc[13202]: __weak variable at 0x7fcaf2eedef0 holds 0x150 instead of 0x7fcaebc14840. This is probably incorrect use of objc_storeWeak() and objc_loadWeak(). Break on objc_weak_error to debug.

objc[13202]: __weak variable at 0x7fcaf3305360 holds 0x7fcaf332c000 instead of 0x7fcaebc14840. This is probably incorrect use of objc_storeWeak() and objc_loadWeak(). Break on objc_weak_error to debug.

2018-07-26 17:29:56.264 Etcher[13202:2853911] -[_NSImageViewSimpleImageView setDoubleValue:]: unrecognized selector sent to instance 0x7fcaf19b1480
Illegal instruction: 4```

Crashes when it's about to ask for permission admin permission via the Modal to get direct read / write access to the device.

jhermsmeier, 2018-07-26:
@Dygear and that's also on the Mojave Beta?

stefanskotte, 2018-07-27:
Same for me too, switched to PI-Baker for now as it still works fine.

i7clock, 2018-07-31:
Same issue when I click FLASH ... also running 10.14 Beta (18A336e)

BigBallerBrand2, 2018-07-31:
Same here as well running 10.14 Beta.

areis422, 2018-08-02:
Same thing here on beta 5

wi-c02wj48mhtdf:~ 20578$ systemstats

Summary
=======
           System Version:	18A336e
           Hardware Model:	MacBookPro14,3
                 Board ID:	Mac-551B86E5744E2388
    Installed Memory Size:	16GB (17179869184)
                Page Size:	4096
         Root Volume Type:	SSD:1,HDD:0,FUSION:0,FDE:1,RAID:0,TINYSSD:0,IN:1,EX:0,INEX:0,APFS
         Root Volume Size:	465GB (499963170816)
                 SSD Name:	APPLE SSD SM0512L
             SSD Revision:	CXS5EA0Q
     Discrete GPU Present:	1
               Total Time:	68:06:00

```
Process:               Etcher [11381]
Path:                  /Applications/Etcher.app/Contents/MacOS/Etcher
Identifier:            io.resin.etcher
Version:               1.4.4 (1.4.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Etcher [11381]
User ID:               1934014795

Date/Time:             2018-08-02 14:37:44.581 -0500
OS Version:            Mac OS X 10.14 (18A347e)
Report Version:        12
Bridge OS Version:     3.0 (14Y665)
Anonymous UUID:        49791D4F-CDB4-B8E3-30DF-290EF82D7C4A

Time Awake Since Boot: 18000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [11381]

Application Specific Information:
Crashing on exception: -[_NSImageViewSimpleImageView setDoubleValue:]: unrecognized selector sent to instance 0x600003cbd860

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff3564f90d __exceptionPreprocess + 256
1   libobjc.A.dylib                     0x00007fff61a0f720 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff356d7a36 -[NSObject(NSObject) __retain_OA] + 0
3   CoreFoundation                      0x00007fff355df290 ___forwarding___ + 1486
4   CoreFoundation                      0x00007fff355dec38 _CF_forwarding_prep_0 + 120
5   Electron Framework                  0x00000001005925ae _ZN4atom15NativeWindowMac14SetProgressBarEdNS_12NativeWindow13ProgressStateE + 606
6   Electron Framework                  0x0000000100564851 _ZN4atom3api6Window14SetProgressBarEdPN4mate9ArgumentsE + 433
7   Electron Framework                  0x00000001005684f5 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 149
8   Electron Framework                  0x00000001005684cc _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 108
9   Electron Framework                  0x00000001005684a2 _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 66
10  Electron Framework                  0x000000010056847d _ZN4base8internal7InvokerINS0_9BindStateIMN4atom3api6WindowEFvdPN4mate9ArgumentsEEJEEEFvPS5_dS8_EE3RunEPNS0_13BindStateBaseEOSC_OdOS8_ + 29
11  Electron Framework                  0x00000001005683fd _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 589
12  Electron Framework                  0x0000000100568351 _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 417
13  Electron Framework                  0x000000010056822c _ZN4mate8internal10DispatcherIFvPN4atom3api6WindowEdPNS_9ArgumentsEEE18DispatchToCallbackERKN2v820FunctionCallbackInfoINSA_5ValueEEE + 124
14  libnode.dylib                       0x0000000104ddbf21 _ZN2v88internal6StrDupEPKc + 609
15  libnode.dylib                       0x0000000104e90ce8 _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 4136
16  libnode.dylib                       0x0000000104e9029b _ZNK2v88internal8compiler17BranchElimination21ControlPathConditionseqERKS3_ + 1499
17  ???                                 0x00000d9daf10437d 0x0 + 14970898105213
[... 887 lines cut ...]
IOKit                             15.5M        2 
Image IO                           256K        2 
Kernel Alloc Once                    8K        2 
MALLOC                           244.6M       36 
MALLOC guard page                   32K        9 
MALLOC_NANO (reserved)           384.0M        2         reserved VM address space (unallocated)
Memory Tag 242                      12K        2 
Memory Tag 251                      76K        3 
Memory Tag 255                   525.0M       58 
STACK GUARD                       56.1M       37 
Stack                            191.8M       37 
VM_ALLOCATE                       10.3M       25 
VM_ALLOCATE (reserved)             256K        2         reserved VM address space (unallocated)
__DATA                            44.1M      355 
__FONT_DATA                          4K        2 
__LINKEDIT                       221.1M       14 
__TEXT                           332.2M      353 
__UNICODE                          560K        2 
mapped file                       75.3M       56 
shared memory                      744K       22 
===========                     =======  ======= 
TOTAL                              2.1G     1165 
TOTAL, minus reserved VM space     1.7G     1165 

```

uchuugaka, 2018-08-04:
Same here on that and later.

anlgrses, 2018-08-06:
Same issue with Mojave 18A347e

kubanored, 2018-08-08:
same issue

MelvinTo, 2018-08-08:
same issue

seenashafai, 2018-08-10:
same issue

sashimiman, 2018-08-11:
Same issue on Mojave PB5

uchuugaka, 2018-08-11:
I have it on good authority it will be resolved in a near future beta build. Sit tight.

chungonion, 2018-08-14:
Just tested with Mojave Develop Beta 7 (Public Beta 6), it works again!
Thanks @uchuugaka for the information above! Cheers! 🍻
[image]

dmix, 2018-08-20:
Yep, worked for me with Public Beta 6. It's likely safe to close this one...

jhermsmeier, 2018-08-27:
Thanks all! Closing, as reported working in macOS Mojave Public Beta 6

closed 2018-08-27 by jhermsmeier