balena-io/etcher #2406: Etcher trying to write C drive MBR

Deej0406, 2018-06-24:
- **Etcher version:** V1.4.4
- **Operating system and architecture:** W10  64b Intel  i5-4200
- **Image flashed:** 32Gb Micro SD
- **Do you see any meaningful error information in the DevTools?**  NO

Version = 1.4.4, Type = nsis
gui.js:105081 Sun Jun 24 2018 21:47:51 GMT+0200 (W. Europe Daylight Time) Application start ({"packageType":"nsis","version":"1.4.4"})
gui.js:105081 Sun Jun 24 2018 21:47:51 GMT+0200 (W. Europe Daylight Time) Checking for updates ({"currentVersion":"1.4.4","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
gui.js:105081 etcher:sdk:scanner start +278ms true
gui.js:105081 Sun Jun 24 2018 21:47:53 GMT+0200 (W. Europe Daylight Time) [object Event]
gui.js:105081 Sun Jun 24 2018 21:47:53 GMT+0200 (W. Europe Daylight Time) Update notification skipped ({"reason":"Latest version"})
gui.js:105081 Sun Jun 24 2018 21:47:56 GMT+0200 (W. Europe Daylight Time) Open image selector
gui.js:105081 Sun Jun 24 2018 21:49:26 GMT+0200 (W. Europe Daylight Time) Select image ({"path":"C:\\Users\\Dave\\Desktop\\Desktop folder\\Downloads\\naturewatch-cam-v0p2p5.img","extension":"img","size":{"original":10393485312,"final":{"estimation":false,"value":10393485312}},"hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93597,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":20299776,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
gui.js:105081 etcher:sdk:scanner stop +2m true
gui.js:105081 Sun Jun 24 2018 21:49:32 GMT+0200 (W. Europe Daylight Time) Flashing 0 devices, 0% at 0 MB/s (total 0 MB/s) eta in undefineds with 0 failed devices
gui.js:105081 Sun Jun 24 2018 21:49:32 GMT+0200 (W. Europe Daylight Time) Flash ({"image":"C:\\Users\\Dave\\Desktop\\Desktop folder\\Downloads\\naturewatch-cam-v0p2p5.img","drives":["\\\\.\\PhysicalDrive1"],"uuid":"a4183405-ee87-4685-91f6-23a1d6162f58","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Elevating command: C:\Users\Dave\AppData\Local\Programs\etcher\Etcher.exe C:\Users\Dave\AppData\Local\Programs\etcher\resources\app.asar
gui.js:105081 Successfully connected to IPC server: etcher-server-3768, socket root C:\Users\Dave\AppData\Local\Temp\
gui.js:105081 Image: C:\Users\Dave\Desktop\Desktop folder\Downloads\naturewatch-cam-v0p2p5.img
gui.js:105081 Devices: \\.\PhysicalDrive1
gui.js:105081 Umount on success: true
gui.js:105081 Validate on success: true
gui.js:105081 Fail: Object
gui.js:105081 Error: EPERM: operation not permitted, write
gui.js:105081 Sun Jun 24 2018 21:49:57 GMT+0200 (W. Europe Daylight Time) Flash error ({"error":{"name":"Error","message":"EPERM: operation not permitted, write","stack":"Error: EPERM: operation not permitted, write","code":"EPERM","syscall":"write","errno":-4048},"image":"C:\\Users\\Dave\\Desktop\\Desktop folder\\Downloads\\naturewatch-cam-v0p2p5.img","drives":["\\\\.\\PhysicalDrive1"],"uuid":"a4183405-ee87-4685-91f6-23a1d6162f58","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Sun Jun 24 2018 21:49:57 GMT+0200 (W. Europe Daylight Time) Open modal ({"name":"warning"})
gui.js:105081 {"stack":"Error: EPERM: operation not permitted, write","message":"EPERM: operation not permitted, write","name":"Error","code":"EPERM","syscall":"write","errno":-4048}
console.(anonymous function) @ gui.js:105081
gui.js:105081 etcher:sdk:scanner start +3m true

I am trying to to set up a Raspberry Pi Zero from my windows laptop onto a micro SD.   Etcher identifies the image and the SD card, but after clicking Flash my acronis software warns me that Etcher is trying to update my C drive MBR record

Digital-Thor, 2019-01-05:
I'm getting a warning as well from Acronis saying "**Possible Ransomware Detected**... tried modifying your MBR"
System: Windows 10
BalenaEtcher: v1.4.9
Drive: 32GB Micro SD

kb3tbx, 2019-01-24:
Confirmed issue here as well:
"Possible ransomware detected
Acronis Active Protection paused the program that tried modifying your MBR."
I blocked it, to add to the blacklist. Then Uninstalled it to find another way.

Windows 10 Intel Core i7-6600U CPU 2.6Ghz Laptop
SanDisk Cruzer Dial USB Device 31.33 GB D:\
Etcher Download for Windows x64 from here: https://www.balena.io/etcher/
I could not find a version number anywhere in the app.

The 1/23/2019 download filename is balenaEtcher-Setup-1.4.9-x64

Bye.

lurch, 2019-01-24:
@Deej0406 These lines from your console log:
```
gui.js:105081 Image: C:\Users\Dave\Desktop\Desktop folder\Downloads\naturewatch-cam-v0p2p5.img
gui.js:105081 Devices: \\.\PhysicalDrive1
```
make it indeed look like you're trying to write the image file to your C: drive (IIRC `\\.\PhysicalDrive1` usually contains the `C:` partition?). Did you select the wrong drive in the drive-selection window by accident?