balena-io/etcher #2407: How To use Cammand use - Formating corrupted micro SD card

kuamit5516, 2018-06-26:
root@kali:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0 651.9G  0 part /
├─sda2        8:2    0     1K  0 part 
└─sda5        8:5    0 279.6G  0 part [SWAP]
sr0          11:0    1  1024M  0 rom  
mmcblk1     179:0    0  14.8G  0 disk 
└─mmcblk1p1 179:1    0  14.7G  0 part /media/root/STRONTIUM .

**My sd card working 100% but why not delete any file and data &  not a format im try format PC & Android use but not format sd card also show error - file not delete filed** 

help me ...

lurch, 2018-06-27:
I'm afraid I don't understand your question. Does https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives help?

kuamit5516, 2018-06-29:
My memorry card working   

 .... see

root@kali:~# lsblk
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
sda 8:0 0 931.5G 0 disk
├─sda1 8:1 0 651.9G 0 part /
├─sda2 8:2 0 1K 0 part
└─sda5 8:5 0 279.6G 0 part [SWAP]
sr0 11:0 1 1024M 0 rom
mmcblk1 179:0 0 14.8G 0 disk
└─mmcblk1p1 179:1 0 14.7G 0 part /media/root/STRONTIUM .

### My Question is my memorry card not format i try agin agin (Im try sd card format by. Pc & laptop but not format also show error this - : - filed delete file )

**i don't understudand why show error  ..?**

### **help me ..**

lurch, 2018-06-30:
How does this relate to Etcher?

kriekprinsloo, 2018-07-01:
Yes, are you having trouble formatting cards in general, or is this an
Etcher specific problem?

kuamit5516, 2018-07-04:
#lurch 

im use this sd card in my android Lava iris405 rooted and im install linux then im format any data my memory card but not delete file ..

why ?? 

help me

jhermsmeier, 2018-07-05:
@kuamit5516 closing this, as this does not appear to be an issue with Etcher from what we can tell. If this *is* a problem with Etcher, please state in detail what the problem is, and how we might reproduce it.

closed 2018-07-05 by jhermsmeier