balena-io/etcher #2415: Manifest file for reproducible builds

ghisvail, 2018-07-08:
Please consider publishing a corresponding manifest file for `etcher` (either a `package-lock.json` or `yarn.lock` file) to enable reproducible builds by downstream packaging solutions, such as [flatpak](https://flatpak.org/) for instance.

Thanks.

lurch, 2018-07-08:
You mean like https://github.com/resin-io/etcher/blob/master/npm-shrinkwrap.json ?
(apologies if I've grabbed the wrong end of the stick!)

ghisvail, 2018-07-08:
My bad, I forgot about that one. Not used to this variety of lock files in the Javascript ecosystem.

ghisvail, 2018-07-08:
I am getting the following warning though: `npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but npm-shrinkwrap.json was generated for lockfileVersion@0. I'll try to do my best with it!`.

And the flatpak helper script for npm fails to parse the manifest (probably because it's only been tested with a more recent `lockfileVersion`).

So far I have only dealt with projects publishing a `yarn.lock` or `package-lock.json`. Would you consider at least upgrading the `npm-shrinkwrap.json` file, or submitting an additional `package-lock.json`. I am not sure I understand the difference between both, and read conflicting explanation after a quick search.

lurch, 2018-07-08:
I wonder if things have "improved" since #1475 or #1487 ? :man_shrugging: 

See also #1528

ghisvail, 2018-07-19:
Actually, I might be able to use `npm-shrinkwrap.json`. However, the tooling is only compatible with v1+ of the lock files, whilst yours is on the old v0 format. Would it be possible to upgrade the lock file format of `npm-shrinkwrap.json` at least?

jhermsmeier, 2018-07-19:
@ghisvail due to a variety of issues with newer npm versions (4, and 5) we're not going to upgrade the shrinkwrap just yet – we still have to look at npm v6, and see if it's fixed all the issues we were having. If you just want to build Etcher, and not add any dependencies, using a newer npm version *should* work fine, even though it warns about the shrinkwrap version. To avoid this entirely, you can also drop npm to v3.10.10 (i.e. switch to Node v6 with your Node version manager of choice).

ghisvail, 2018-07-30:
@jhermsmeier The issue is not with the build itself, but with `lockfileVersion 0` which is not supported by the `flatpak` suite. I am fine with keep `npm-shrinkwrap.json` as it is, however, do you know how I can upgrade it to `lockfileVersion 1` locally?

jhermsmeier, 2018-07-31:
> but with lockfileVersion 0 which is not supported by the flatpak suite

I'm confused – what does the shrinkwrap's lockfileVersion have to do with that (and what exactly do you mean by the flatpak suite)?

> however, do you know how I can upgrade it to lockfileVersion 1 locally?

Yes, you can delete the `npm-shrinkwrap.json` and then run `npm shrinkwrap` with the latest `npm`, which will create a shrinkwrap file with a lockfileVersion of 1

ghisvail, 2018-08-01:
> ou can delete the `npm-shrinkwrap.json` and then run `npm shrinkwrap` with the latest `npm`

All I get it the following:

```json
{
  "name": "etcher",
  "version": "1.4.4",
  "lockfileVersion": 1
}
```

None of the dependencies are listed.

jhermsmeier, 2018-08-01:
I should've mentioned that you need to install the dependencies before deleting that & re-shrinkwrapping. You can do that with `make electron-develop` (which ensures that native bindings are built against the Electron runtime)

closed 2018-10-15 by jviotti

ghisvail, 2018-10-24:
All good, thanks for your help @jhermsmeier.