balena-io/etcher #2417: Will not open

ZWannamaker, 2018-07-15:
I just downloaded etcher from etcher.io today. Ran the setup and tried to open the software. Nothing happens. Redownloaded several times and reinstalled several times and restarted server times no luck.

jhermsmeier, 2018-07-15:
Which operating system are you installing Etcher on?
Which installer did you download?
Which version of Etcher are you installing?

Without further information, we can only guess. Answering the questions from the issue template would greatly reduce us having to ask these same questions again.

ZWannamaker, 2018-07-15:
windows 10
windows x64
v1.4.4

ZWannamaker, 2018-07-15:
Just tried the last 5 versions and same results.

luke99k23, 2018-07-18:
I'm trying to make a rasberry pie zero in a NES case and I am attempting to follow the directions for etcher. It is not allowing me to pass the 'select drive phase' because it states that the 'drive contains image'. I already left it zipped and unzipped. I do not know what to do next to get past this part in the process for my SD card.  Help!

luke99k23, 2018-07-18:
https://howchoo.com/g/n2qyzdk5zdm/build-your-own-raspberry-pi-retro-gaming-rig

luke99k23, 2018-07-18:
The link above is what I am attempting to follow.

lurch, 2018-07-19:
"drive contains image" means that the drive you're trying to write the image **to** is also the same drive that you're trying to read the image **from**. This obviously won't work - try copying the image to e.g. your computer's hard drive, and then writing it to your SD card.

luke99k23, 2018-07-19:
Thank you. That helped and now I feel very foolish. :) I will probably need more help soon.

lurch, 2018-07-19:
@luke99k23 For any future problems, it'd be better to open a new issue, rather than comment on an unrelated issue. Or you could try asking on https://forums.resin.io/c/etcher

holgersprute, 2018-07-29:
Was the original issue solved? I have the very same issue, same setup. Restart did not help either, same issue on portable version.

closed 2018-10-15 by jviotti