balena-io/etcher #2419: Errors spewing on the screen when flashing NEC screen

vpetersson, 2018-07-20:
- **Etcher version:**  v1.4.4
- **Operating system and architecture:**  Windows 10 / amd64(?)
- **Image flashed:** Screenly Pro 

I didn't flash this myself, but was on the phone when this happened. In short, the flashing worked fine when flashing an NEC screen, but a flood of messages were displaying on the screen:

[image]

Unsure if this is expected, but would probably make sense to suppress these errors if it's expected.

jhermsmeier, 2018-07-20:
Hey @vpetersson, that looks more like a fault with the OS or the hardware to me, rather than an issue with Etcher. Or has this worked before with the same, or different version of Etcher and the same OS image?

vpetersson, 2018-07-20:
@jhermsmeier I don't think that's the OS. There is no OS running on it at this point as it was corrupted. Moreover, it was picked up by Etcher and we were able to successfully flash it at "enhanced speed" without `rpiboot`. Hence, my hypothesis would be that it's related to the OS that Etcher sends to the Pi for improved flash performance.

jhermsmeier, 2018-07-20:
The log output you see on the screen you posted here is from the Linux kernel, and the `dwc2_hsotg` is part of a USB OTG driver, which makes me think it might have been an issue with the OS, or attached hardware.

Can you give us a little more detail around what board you're flashing and how? You mentioned that the image that caused this was flashed with `rpiboot`, and that it works when flashing the storage (SD-card?) directly, correct?

Further, is this something that is reproducible, e.g. does this always happen when you flash it as you did when this occurred?

vpetersson, 2018-07-20:
Oh, sorry. I should have given you more background here.

 * This was while flashing an NEC P or V series monitor
 * These screens comes with a [built-in](https://www.necdisplay.com/raspberry-pi/) Raspberry Pi Compute Module 3 (modified to have 16GB of storage)
 * Given that it's a CM3, there is no SD card - it's all using the built-in eMMC

I unfortunately do not have access to the monitor myself, so I cannot try to reproduce it. What I can confirm however is that the device worked just fine and that the flashing worked (just power it off and when powering it back on, it worked fine with the new image).

It is my understanding that Etcher will use USB boot to send a custom image to the device when flashing a CM3 in order to improve the write performance (@alexandrosm please correct me if I'm wrong). Hence my guess is that either nobody tested flashing out a CM3 (or Fin) with a display connected, or that this is specific to the add-on board for NEC uses.

jhermsmeier, 2018-07-20:
Oh, so this log that shows up on screen is *while* flashing the device via usbboot, if I understand this correctly now?

> the flashing worked (just power it off and when powering it back on, it worked fine with the new image)

That's expected, as you have to boot into the flashed image (and during usbboot flashing the kernel that Etcher sends over is running), so that power-cycling (or rebooting) is necessary.

Since the usbboot kernel is just a tiny kernel, with only the parts needed to talk USB & write to the builtin flash memory (iirc), the console output would be expected, as it doesn't start a desktop environment or anything really.

vpetersson, 2018-07-20:
> Oh, so this log that shows up on screen is while flashing the device via usbboot, if I understand this correctly now?

Yep! That's right.

> That's expected, as you have to boot into the flashed image (and during usbboot flashing the kernel that Etcher sends over is running), so that power-cycling (or rebooting) is necessary.

Yes, that part is expected.

> Since the usbboot kernel is just a tiny kernel, with only the parts needed to talk USB & write to the builtin flash memory (iirc), the console output would be expected, as it doesn't start a desktop environment or anything really.

Right, of course. My "issue" so to speak was rather that it would make sense to suppress the output as it made the non-technical people on the other side think it was broken.

What I would suggest would be to either just show a splash screen (perhaps with [psplash](https://git.yoctoproject.org/cgit/cgit.cgi/psplash)) or to suppress the logging.

jhermsmeier, 2018-07-20:
I see now! Took me a moment to understand :) I think this would be a good thing indeed, and have added https://github.com/resin-io-modules/node-raspberrypi-usbboot/issues/12 to track this (as that's the module Etcher will be using for usbboot in the future). Thanks for bringing this up!

vpetersson, 2018-07-20:
Haha yeah, lots of moving parts here ;)

vpetersson, 2018-07-20:
Btw - Would be awesome to have a feature in Etcher to reboot the Pi after the flashing (perhaps a dialogue since it isn't always desirable).

jhermsmeier, 2018-07-20:
Yeah, I was thinking in that direction as well, as the power-cycling came up; would you fancy adding that as an issue to [resin-io-modules/node-raspberrypi-usbboot](https://github.com/resin-io-modules/node-raspberrypi-usbboot)?

vpetersson, 2018-07-20:
Sure thing.

closed 2018-10-15 by jviotti