balena-io/etcher #2421: Etcher-cli: after flasing image, the drive becomes unaccessible

SuperJMN, 2018-07-20:
- **Etcher version:** 
1.1.2
- **Operating system and architecture:** 
Windows 10 (1803) x64 
- **Image flashed:**
Gpt.zip => https://github.com/andreiw/RaspberryPiPkg/files/2111547/gpt.zip 
- **Do you see any meaningful error information in the DevTools?** 
No error

The problem I'm observing is that after the image is flashed, the flashed disk becomes inaccessible. In order to be accessible again, I have to unplug and replug the thumbdrive. It doesn't even appear in the Safe Eject menu (it shows "---" instead).

**This is a big problem**, because it really limits the usage from a script. It would require the user to unplug-replug in the middle of the script. You know how inconvenient this is.

Is there any way to make etcher

jhermsmeier, 2018-07-20:
@SuperJMN have you tried the `--no-unmount` flag?

SuperJMN, 2018-07-20:
No. Will try ASAP!

SuperJMN, 2018-07-20:
OK, I've tried it and it works perfectly! What drives me to a question: why isn't this the default?

closed 2018-07-20 by SuperJMN

jhermsmeier, 2018-07-20:
Glad that worked out! This isn't the default, because people typically want to unplug the devices after flashing, without having to unmount them manually first.