balena-io/etcher #2424: "Unplugged accidentally"?

trendol, 2018-07-22:
Hey. Etcher has worked perfectly til now. Running latest version of Etcher, Mac OS X 10.13.3. It writes an RPI image to my SD card perfectly, but when validating it's stuck on 35-36% and after 5-6 minutes it returns a

"Looks like Etcher lost access to the drive.
Did it get unplugged accidentally?

Sometimes this error is caused by faulty reders that don't provide stable access to the drive."

..and a Retry-button.

The SD-card reader is not faulty, since I've used it with both read/write for alot of other cards.

Not sure what's going on here - help?

lurch, 2018-07-24:
I wonder if this might be another variant of #1416 ?

trendol, 2018-07-24:
I’ll give it a try, thanks!

jhermsmeier, 2018-07-24:
I don't think so @lurch, that'd only happen *after* verification, while this looks like something odd is happening in the middle of it.

@trendol does this keep happening with that combination of image & sd-card? And did your mac go into sleep, or were you observing the entire process?
Also, can you tell us the exact version number of Etcher you're running; "latest" will mean different things later in time (and some people also miss updates, etc.), which makes it rather inaccurate ;)

closed 2018-10-15 by jviotti

pattagobi, 2018-10-23:
@jhermsmeier hello, I don't have Mac OS, I have windows 10, 
etcher version is 1.4.5, 
 the etcher just stops in between at "flashing".
and as I have a desktop on performance mode so no sleeping of anything.
and yes I was observing it while it was doing it stuff
and it is happening in my 3 different sd cards, class 4, and 2 class 10.
tried different images too, Raspbian lite, noobs, stretch, windows IoT. Nothing worked.

and i guess my cards are working fine as i  tested on phone and tried to transfer data just to be sure

please help.

kriekprinsloo, 2018-10-24:
What I would try is to reinstall etcher, and try an external SD reader (or
if you are already using one try a different one) to be safe.

pattagobi, 2018-10-24:
@kriekprinsloo tried sd reader too.
got fed up. 
succeeded with win32 disk imager.