balena-io/etcher #2426: Debian 9 Stretch gpg key issue

ralph-de-neuton, 2018-07-30:
- **Etcher version:**  n/a
- **Operating system and architecture:** amd64 Debian 9 Stretch 
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?**  n/a

following the instructions on installing with .deb file using Debian/Ubuntu gave me this response:

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.hNYXKwsToa/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: failed to start the dirmngr '/usr/bin/dirmngr': No such file or directory
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.hNYXKwsToa/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr

lurch, 2018-07-30:
Does this help? https://github.com/resin-io/etcher/issues/2088

jhermsmeier, 2018-07-31:
@ralph-de-neuton as the error message states, `dirmngr` is missing from your system; you can install it via apt:

```
sudo apt-get install dirmngr
```

closed 2018-10-15 by jviotti