balena-io/etcher #2429: Class 10 Sandisk microSD memory card not detecting

digen, 2018-08-01:
- **Etcher version: 1.4.4** 
- **Operating system and architecture: Mac OSX 10.10.5** 
- **Image flashed: N/A** 
- **Do you see any meaningful error information in the DevTools? N/A** 

Hello,
Thank you for making Etcher !
 Tried a couple of brand new Sandisk Class 10 microSD cards. Won't detect in 1.4.4. 
Downgraded to 1.3.1 as the .img was already there on the system and that version detected perfectly. 

Is there any other information I can provide to help you debugging this issue ?

lurch, 2018-08-01:
Sounds like maybe another https://github.com/resin-io-modules/drivelist bug? Does your SD card show up if you enable "Unsafe mode" in the settings?

jhermsmeier, 2018-08-01:
Yup, looks like https://github.com/resin-io-modules/drivelist/issues/292

lurch, 2018-08-01:
...except https://github.com/resin-io-modules/drivelist/issues/292 concerns Linux, and @digen has said he's using Mac OSX?

jhermsmeier, 2018-08-02:
Right, wrong issue – let me find the correct one

jhermsmeier, 2018-08-02:
This is the one: https://github.com/resin-io/etcher/issues/2357

jhermsmeier, 2018-08-02:
Duplicate of https://github.com/resin-io/etcher/issues/2357

jhermsmeier, 2018-08-02:
We're already working on this, I'll mark this as a dupe, and close it.

closed 2018-08-02 by jhermsmeier