balena-io/etcher #2438: Silent Install via SCCM2012

cyelvington, 2018-08-06:
- **Etcher version: 1.4.4** 
- **Operating system and architecture: Windows x64** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

I was referred here via the Resin.io facebook page -- I am working for a university and need to deploy Etcher to Windows machines via SCCM 2012. Is there a silent installer option for Etcher, and if so, what is the switch I need to use?

Thanks in advance!

--Clinton

lurch, 2018-08-06:
Does https://github.com/resin-io/etcher/blob/master/README.md#chocolatey-windows allow you to do what you want?

jhermsmeier, 2018-08-28:
You should be able to run it silently with the `/S` flag, as detailed in [NSIS' installer usage](http://nsis.sourceforge.net/Docs/Chapter3.html#installerusage), as the Etcher installer is built with NSIS

closed 2018-08-28 by jhermsmeier