balena-io/etcher #2446: Drives always locked in Ubuntu, but not in windows

onokje, 2018-08-12:
- **Etcher version: Version = 1.4.4, Type = appimage** 
- **Operating system and architecture: Ubuntu 16** 
- **Image flashed: Any** 
- **Do you see any meaningful error information in the DevTools? No** 

I can't use this tool in Ubuntu. Any SD card I try, or any image, it always says the SD card is Locked in the Select drive modal (the red "Locked" label). I checked if the drive is physically locked: It is not. If i switch to windows 10 on the same PC (I have dualboot) and use the windows version, it works fine. So i have a workaround for myself (switch to Windows), but it is still kinda annoying.

lurch, 2018-08-14:
@jhermsmeier Another `drivelist` bug?

jhermsmeier, 2018-08-28:
@onokje what are those SD cards formatted as? Do they only have one partition, or several? Are they automatically mounted on your Ubuntu?

jhermsmeier, 2018-08-28:
Thanks @onokje! Let us know how it goes

jhermsmeier, 2018-08-28:
Also, if you could post the output of running `lsblk --bytes --all --json --paths --output-all` in a terminal, that'd be very helpful!

closed 2018-10-15 by jviotti

onokje, 2019-08-03:
The problem still exists on ubuntu 18.

output of lsblk:

`{
   "blockdevices": [
      {"name": "/dev/loop0", "kname": "/dev/loop0", "maj:min": "7:0", "fstype": "squashfs", "mountpoint": "/snap/gnome-logs/45", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "15208448", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop1", "kname": "/dev/loop1", "maj:min": "7:1", "fstype": "squashfs", "mountpoint": "/snap/gnome-characters/296", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "15462400", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop2", "kname": "/dev/loop2", "maj:min": "7:2", "fstype": "squashfs", "mountpoint": "/snap/gnome-logs/61", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "1032192", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop3", "kname": "/dev/loop3", "maj:min": "7:3", "fstype": "squashfs", "mountpoint": "/snap/gnome-3-28-1804/67", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "157184000", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop4", "kname": "/dev/loop4", "maj:min": "7:4", "fstype": "squashfs", "mountpoint": "/snap/gnome-system-monitor/100", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "3825664", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop5", "kname": "/dev/loop5", "maj:min": "7:5", "fstype": "squashfs", "mountpoint": "/snap/gtk-common-themes/1198", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "37027840", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "0", "pkname": null, "hctl": null, "tran": null, "subsystems": "block", "rev": null, "vendor": null, "zoned": "none"},
      {"name": "/dev/loop6", "kname": "/dev/loop6", "maj:min": "7:6", "fstype": "squashfs", "mountpoint": "/snap/core/6964", "label": null, "uuid": null, "parttype": null, "partlabel": null, "partuuid": null, "partflags": null, "ra": "128", "ro": "1", "rm": "0", "hotplug": "0", "model": null, "serial": null, "size": "92733440", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "none", "rq-size": "128", "type": "loop", "disc-aln": "0", "disc-gran": "4096", "disc-max": "4294966784", "disc-zero": "0", "wsame": [... cut ...] 06733824", "state": null, "owner": "root", "group": "disk", "mode": "brw-rw----", "alignment": "0", "min-io": "512", "opt-io": "0", "phy-sec": "512", "log-sec": "512", "rota": "1", "sched": "cfq", "rq-size": "128", "type": "part", "disc-aln": "0", "disc-gran": "0", "disc-max": "0", "disc-zero": "0", "wsame": "0", "wwn": null, "rand": "1", "pkname": "/dev/sdb", "hctl": null, "tran": null, "subsystems": "block:scsi:usb:pci", "rev": null, "vendor": null, "zoned": "none"}
         ]
      }
   ]
}
`

thundron, 2019-08-05:
@onokje Which version of Etcher are you running? Have you tried also with the latest available?

onokje, 2019-08-05:
1.5.52 (AppImage)