balena-io/etcher #2451: Cannot find module 'lzma-native' when running `make test-cli`

jedahan, 2018-08-17:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** macOS 10.13.6
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?**  n/a

When building etcher-cli via `make test-cli` I get an error:

```
...build stuff...
mocha --recursive --reporter spec \
                tests/shared \
                tests/image-stream
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'lzma-native'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/Volumes/data/code/etcher/lib/sdk/image-stream/handlers.js:24:36)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/Volumes/data/code/etcher/lib/sdk/image-stream/index.js:24:18)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/Volumes/data/code/etcher/lib/shared/supported-formats.js:21:21)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/Volumes/data/code/etcher/tests/shared/supported-formats.spec.js:21:26)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at /Volumes/data/code/etcher/node_modules/mocha/lib/mocha.js:231:27
    at Array.forEach (<anonymous>)
    at Mocha.loadFiles (/Volumes/data/code/etcher/node_modules/mocha/lib/mocha.js:228:14)
    at Mocha.run (/Volumes/data/code/etcher/node_modules/mocha/lib/mocha.js:536:10)
    at Object.<anonymous> (/Volumes/data/code/etcher/node_modules/mocha/bin/_mocha:582:18)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:266:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:596:3)
make: *** [test-cli] Error 1
```

lurch, 2018-08-21:
I think you need to do `make cli-develop` first?
(i.e. this is probably a documentation bug)

jedahan, 2018-08-23:
maybe we can just add cli-develop to the test-cli dep in the makefile?

Should be fine even if it rebuilds every time you test?

```diff
diff --git a/Makefile b/Makefile
index 1e9eb5a1..9c5bd3df 100644
--- a/Makefile
+++ b/Makefile
@@ -571,7 +571,7 @@ test-sdk:
                tests/shared \
                tests/image-stream

-test-cli:
+test-cli: cli-develop
        mocha $(MOCHA_OPTIONS) \
                tests/shared \
                tests/image-stream
```

lurch, 2018-08-23:
P.S. My previous attempt to tidy up the `Makefile` ( #1896 ) got ignored, it's now terribly out of date, and I can't be bothered to rebase it :wink:

closed 2018-10-15 by jviotti