balena-io/etcher #2454: Etcher-cli builds a bunch of unnecessary electron deps

jedahan, 2018-08-19:
- **Etcher version:** 1.4.4
- **Operating system and architecture:**  macOS 10.13.6
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?**  n/a

When building with `make package-cli RELEASE_TYPE=production`, it takes about 30 minutes on my laptop (13" 2012 rMBP), most of which is in electron dependencies.

By applying the patches described in https://github.com/chrmoritz/homebrew-core/blob/cf5e86e9d301d82df04dc6e1d137f69f8bc1d7c2/Formula/etcher-cli.rb , build time drops to about 2 minutes.

The relevant changes that speed up building:

* removing '@fortawesome', '@types', 'angular', 'react', 'redux', 'electron' from dependencies
* removing devDependencies
* setting files to be bin, build, lib/cli, lib/sdk, lib/shared, and binding.gyp
* getting rid of npm-shrinkwrap so these changes to get overridden

Of course this only works for etcher-cli. I'm not familiar with Makefiles or how this could be officially supported, but is there a way?

lurch, 2018-08-21:
I guess to do this "officially" there'd need to be different `package.json` files for the GUI and CLI? I doubt that's something the Etcher developers are interested in though.

jedahan, 2018-08-23:
Maybe it would be better to rename this issue or create a separate issue to split out etcher-cli and have etcher depend on etcher-cli?

lurch, 2018-08-23:
:man_shrugging: 
You might also want to take a look at #2403

jedahan, 2019-03-05:
Looks like the etcher gui is based on etcher-sdk now, so re-opening the discussion. What's the best way to install & test etcher-cli?

jedahan, 2020-12-05:
Looks like etcher-cli was removed, anyone have other suggestions for a simple commandline image writer?

closed 2020-12-05 by jedahan