balena-io/etcher #2455: Supporting building on system node

jedahan, 2018-08-19:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** macOS 10.13.6
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** n/a

It would be nice if the Makefile supported building and running on system node, instead of forcing downloading node v6.

At the very least it would be nice if we could override the node version using Makefile vars to v8 or v10 without having to edit the Makefile.

What would need to happen to support this?

lurch, 2018-08-21:
See https://github.com/resin-io/etcher/issues/2415#issuecomment-406319392 as a starting point of the problems :wink:

closed 2018-10-15 by jviotti

jedahan, 2018-10-15:
Would be nice to see why this was closed - either that node v8 or up is not supported, or that it is supported.