balena-io/etcher #2456: Blank window on start

shavee, 2018-08-20:
- Etcher version: any on Windows (tried 1.4.4 portable, 1.4.4 x64, 1.4.3 portable, 1.4.3 x64, 1.3.1 portable)
- Windows 10 x64 fully updated
- DevTools not working

Nothing more to explain... this is what it looks like:
[image]

Dev tools (ctrl+shift+I) shows like this:
[image]

Tried running as administrator, running from different units... always the same result.

Any clue?

lurch, 2018-08-20:
I wonder if this is related to #1866 ?

shavee, 2018-08-21:
SOLVED! Same problem as in #696 . My laptop was set to force NVIDIA graphics instead of integrated ones. Switch this to force integrated graphics and Etcher starts without problems.

Thanks!

lurch, 2018-08-21:
@jhermsmeier Maybe something to add to the docs?

closed 2018-10-15 by thundron

koper87, 2019-12-30:
I have an AMD graphic card, so the solution up there didn't work for me. I was facing the same problem until I did change the compatibility mode to Windows Vista, in the Compatibility tab on the executable' Properties menu. 
I hope it helps.

vanhvungve, 2020-01-19:
> I have an AMD graphic card, so the solution up there didn't work for me. I was facing the same problem until I did change the compatibility mode to Windows Vista, in the Compatibility tab on the executable' Properties menu.
> I hope it helps.

well it works

TheOD706, 2020-03-29:
Look like Etcher use component of base browser (maybe Internet Explorer), but you all not use/update IT. I use other OS and it work.

bpp9614, 2020-04-06:
Having same problem Windows, hp laptop. Intel HD graphics 620. Blank grey screen on running npm start.

kodouglass, 2020-05-20:
worked when using internet explorer, failed with chrome

poshest, 2020-06-02:
Same problem, Windows 7, Intel HD Graphics 3000. v1.5.86 seems to work in Vista compatibility mode (although etching fails), but I can't get v1.5.95 working at all.

imarindam, 2020-07-03:
Same problem with Windows 10, AMD CPU and GPU.

thundron, 2020-07-08:
@imarindam can you try updating the .NET framework? if that doesn't work, another thing to try is running from `C:` in case you're using the portable version.
You might also try downgrading and running with compatibility mode as previously mentioned by @poshest.
This is probably not related to Etcher itself so I wouldn't know how to proceed if none of these options work

riccardosimo, 2022-05-12:
same here with the 1.7.9 version