balena-io/etcher #2458: can't install on ubuntu 18.4

Netoperz, 2018-08-27:
- **Etcher version:** 
Actual from repo

- **Operating system and architecture:** 
Linux n37--Stellar 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

- **Image flashed:** 
https://github.com/bamarni/pi64/releases/download/2017-07-31/pi64-lite.zip

- **Do you see any meaningful error information in the DevTools?** 

====
sudo echo "deb https://dl.bintray.com/resin-io/debian stable etcher" | sudo tee /etc/apt/sources.list.d/etcher.list
tee: /etc/apt/sources.list.d/etcher.list: No sucha file or directory

and:

n37@n37--Stellar:~/Pobrane$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61

Error translations:
Niemożna czytac - Cannot read
Nie ma takiego pliku ani katalogu - There is no sucha file or directory

W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
Executing: /tmp/apt-key-gpghome.KBEDw2x3Xw/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: klucz 379CE192D401AB61: klucz publiczny ,,Bintray (by JFrog) <bintray@bintray.com>'' wczytano do zbioru
gpg: Ogółem przetworzonych kluczy: 1
gpg:               dołączono do zbioru: 1
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
W: Nie można czytać /etc/apt/apt.conf.d/ - DirectoryExists (2: Nie ma takiego pliku ani katalogu)
/usr/bin/apt-key: 334: /usr/bin/apt-key: cannot create /etc/apt/trusted.gpg: Directory nonexistent
cat: /tmp/apt-key-gpghome.KBEDw2x3Xw/gpgoutput.log: Nie ma takiego pliku ani katalogu

mDfRg, 2018-09-16:
Here is English version of the same problem: 
Etcher version from repo 
Ubuntu 18.04
Linux HQ 4.15.0-34-generic #37-Ubuntu SMP Mon Aug 27 15:21:48 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

`apt install etcher-electron 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libpango1.0-0 libpangox-1.0-0
The following NEW packages will be installed:
  etcher-electron libpango1.0-0 libpangox-1.0-0
0 upgraded, 3 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/57.2 MB of archives.
After this operation, 220 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Selecting previously unselected package libpangox-1.0-0:amd64.
(Reading database ... 294510 files and directories currently installed.)
Preparing to unpack .../libpangox-1.0-0_0.0.2-5_amd64.deb ...
Unpacking libpangox-1.0-0:amd64 (0.0.2-5) ...
Selecting previously unselected package libpango1.0-0:amd64.
Preparing to unpack .../libpango1.0-0_1.40.14-1ubuntu0.1_amd64.deb ...
Unpacking libpango1.0-0:amd64 (1.40.14-1ubuntu0.1) ...
Selecting previously unselected package etcher-electron.
Preparing to unpack .../etcher-electron_1.4.4_amd64.deb ...
Unpacking etcher-electron (1.4.4) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.1) ...
Processing triggers for bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Setting up libpangox-1.0-0:amd64 (0.0.2-5) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Setting up libpango1.0-0:amd64 (1.40.14-1ubuntu0.1) ...
Setting up etcher-electron (1.4.4) ...
ln: failed to create symbolic link '/usr/local/bin/etcher-electron': No such file or directory
dpkg: error processing package etcher-electron (--configure):
 installed etcher-electron package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Errors were encountered while processing:
 etcher-electron
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

mDfRg, 2018-09-16:
For some strange reason Ubuntu 18.04 doesn't provide path /usr/local/bin.
Manually creating it with
`sudo mkdir -p /usr/local/bin`
solves the problem.

thundron, 2018-12-31:
I don't think this is related to Etcher, is it @jviotti @zvin ?

lurch, 2018-12-31:
According to the [FHS](https://wiki.debian.org/FilesystemHierarchyStandard) `/usr/local/bin` is for "locally compiled binaries, local shell script, etc.". IIRC executables installed by the package manager (i.e. `apt`) ought to go in `/usr/bin` ? Perhaps the `.deb` file is being created with a wrong path somewhere?

@mDfRg 1.4.4 is quite a few versions behind. @thundron have .debs started getting built yet since the "big renaming" ?

thundron, 2018-12-31:
@lurch Yes, we're gonna add the new `.debs` in the repo in the next days so this might become outdated (I hope so at least :P )

closed 2019-05-14 by thundron