balena-io/etcher #2459: no partition selection?

BILLPC2684, 2018-08-29:
Etcher version:
Operating system and architecture: Windows 10 64_86x
Image flashed: ubuntu
Do you see any meaningful error information in the DevTools? no

so i made a 1GB partition on my SD card so i can flash ubuntu on it but it wouldn't let me chose a partition to write to so before i clicked continue i backed up my files and ran with it and yah... (good think i backed up)
and yes it saw there was multiple partitions mounted but just used the whole drive instead...
is there a way for there to be a partition selection?

lurch, 2018-09-03:
Nope, because Etcher works at a raw-disk level "below" partitions, and will always overwrite the partition table at the start of the disk. Sometimes the disk images themselves (that Etcher writes) contain multiple partitions (e.g. Raspberry Pi images).

closed 2018-10-15 by jviotti

raybellis, 2019-06-28:
This would be a *really* useful feature to have.

The scsi2sd SCSI device is very popular among users of vintage samples, etc, and it allows mapping a single SD card into up to 4 (or 6 in the V6 version) virtual SCSI devices.

As such it's recommended by some to map each of those virtual devices to an O/S level partition so that the O/S also sees the same set of virtual disks.   Doing this allows lower level utils such as `dd` to see each virtual disk individually (without worrying about sector offsets) and also allows the O/S to ensure that disk operations cannot over-run the end of one disk onto the start of another.

Without partition level writing Etcher is useless for this use-case, leaving users resorting to much more "unsafe" methods of copying disk images onto their SD cards.