balena-io/etcher #2462: Warn that Etcher deletes all files on the disk

breznak, 2018-09-01:
Ok, I didn't read man pages or docs here, but just found my way that flashing an OS on a flashdrive means deleting all files there. 
I think there should be more proactive warning/documentation of that! Best a dialog if files are found on the selected drive. 
This esp. bites as unetbootin, the tool I've used prior did keep users files. 

Best approaches: 
1. keep files intact, as unetbootin did
2. dialog with warning if there are user files present

lurch, 2018-09-03:
unetbootin works differently to Etcher, see #413

heartbr0ken, 2018-09-09:
I was heartbroken by this as well. I second the kind idea of putting a warning page in the actual deployment of the program.

closed 2018-10-15 by jviotti

breznak, 2018-10-17:
@jviotti  would you care to elaborate why you've just closed the issue? Esp. since it's data-loss related..

jviotti, 2018-10-18:
@breznak This is not something we can easily handle from Etcher's point of view. Etcher is a byte-by-byte flasher, and is expected to wipe all the data and partition table (this is literally the core mission of Etcher).

We could show a dialog if there are files present, but this is not as simple as it sounds. You might be plugging in a drive with a file-system that is not readable in the host operating system and thus don't detect any files even though there might be some.

breznak, 2018-10-19:
> and is expected to wipe all the data and partition table (this is literally the core mission of Etcher). 

agree and it's fine, imho just warn about it. 

> but this is not as simple as it sounds. You might be plugging in a drive with a file-system that is not readable in the host operating system and thus don't detect any files even though there might be some. 

I wouldn't bother with any file-detection code, just add a "Etcher will wipe all the files in the flash drive, continue?" to the reflash dialog and that's it. What do you think?

jviotti, 2018-10-20:
Yeah, that might be worth considering. @thundron @konmouz ?

AnneKir, 2021-02-01:
nice to know this was closed 2 years ago, just lost my all my files, because it was assumed I knew it would delete it.

dennisExploring, 2024-07-12:
> @breznak This is not something we can easily handle from Etcher's point of view. Etcher is a byte-by-byte flasher, and is expected to wipe all the data and partition table (this is literally the [...]

tso dissapoint lost my data

dennisExploring, 2024-07-12:
> nice to know this was closed 2 years ago, just lost my all my files, because it was assumed I knew it would delete it.

same

breznak, 2024-07-12:
It is a shame the issue was just closed (and never reopened), prior to discussion.. where a we found a "compromise" (it is really not that hard to print "Beware, your drive will be fully erased, all data on it will be lost!" )

breznak, 2024-07-12:
@lurch @jviotti can you at least reopen and acknowledge the issue? So someone might care to fix it, thx

lurch, 2024-07-12:
@breznak Neither myself nor @jviotti are working on Etcher any more. If you'd like to see this "fixed", you should probably open a new issue so that the current Etcher maintainers are able to see it.