balena-io/etcher #2464: Etcher destroyed 2 of my USB sticks, but I bet it's not etcher's fault

nxdiamond, 2018-09-02:
Some cheap USB sticks fail when their internal memory chips cannot be written. The internal controller chip switches to read-only mode so that you can rescue files that you had in the stick, and copy them to another drive.

Some cheap USB sticks fail when their internal controller chip cannot communicate with their internal memory chips.  The internal controller chip reports no media.  You cannot rescue your files unless you're the manufacturer or if you have special tools that the manufacturer is supposed to use for their own purposes.

Now, here's my guess about why Rufus and Etcher have a disproportionately large frequency of encountering these errors.

People use Rufus and Etcher to write several gigabytes in a short time.

I bet the internal electronics overheat in cheap USB sticks.

nxdiamond, 2018-09-02:
Last night I figured out a possible workaround, but I have mixed feelings. Application software shouldn't have to cover hardware manufacturers' asses, but innocent bystanders would benefit if their USB sticks don't die. Maybe write chunks of 10MB at a time, and wait 2 seconds after each 10MB.

nxdiamond, 2018-09-03:
Of course there's a problem with fake capacities being written into USB firmware, but that isn't the reason for a stick to fail by reporting "no media" or by converting into read-only mode.

My sticks that failed by reporting "no media" were made by Adata and passed through legitimate wholesale and retail channels.  The retailer accepted returns.

Off topic, but you might notice something if you Google this:
nvidia dell lawsuit
It is not so rare for legitimate manufacturers to learn the hard way that they saved a few pennies by forgetting to test for heat dissipation.

nxdiamond, 2018-09-04:
It took longer to go back to the store, show their employee in one of their PCs that the USB stick was reporting "no media", take the replacement stick back home, and repeat[*], than it would have taken to slow the write speed of Etcher.

As I mentioned, I don't think application software should cover manufacturers' asses, but for end users it might be better to do so for a while.  Let's hope manufacturers will learn what they have to test, but in the meantime it's possible to relieve the burden on end users.

[* Insert old joke about why a programmer starved to death in the shower.  But then observe that in the present situation the word "repeat" might really mean an infinite loop.]

lurch, 2018-09-04:
> than it would have taken to slow the write speed of Etcher.

Feel free to submit a Pull Request... :wink:  (tongue firmly in cheek)

mzs112000, 2018-09-28:
An idea for the tech savy, if you have a few of these cheap USB sticks, open them up(which on it's own will increase air flow), and put a heatsink on the flash chip inside...

closed 2018-10-15 by jviotti

armanit93, 2018-12-27:
be very careful with using this application. It's destroyed 2 USB's from my side as well!!!

rich7, 2024-01-05:
This application Destroyed 2 of my USB sticks. First I thought it was an older usbdrive what was broken, but I bought a new one and the same happened. The USB drive died again. 
On other forums and Reddit you see people complain about this issue where very expensive usb drives and flash cards got lost after using this app;.