balena-io/etcher #2465: Option to wipe data

afriza, 2018-09-03:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** macOS x86_64
- **Image flashed:** https://www.openmptcprouter.com/files/openmptcprouter-v0.34.2-r7288-a662d6f6bc-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz
- **Do you see any meaningful error information in the DevTools?** No Errors. Flashing is successful.

### Steps:
- Flash gzip file to sdcard
- Insert sdcard to Raspberry Pi 3 B+, then set password and do some configurations
- Shutdown RPi, and reflash the same gzip
### Results
The password (SSH and WebUI login) remain there along with the configurations. Flashing the sdcard doesn't wipe data/configurations are not wiped/erased.
### Expected
There is an option to wipe/erase the data and configuration when flashing.
### Workaround
As a workaround for the current behavior, I have to flash another image (e.g. [raspbian lite](https://www.raspberrypi.org/downloads/raspbian/)) file before flashing the gzip file again.

lurch, 2018-09-03:
Sounds like that's probably an "issue" with the image you're flashing. I see that it mentions "sysupgrade" in the filename, so maybe that image is _supposed_ to leave passwords and configurations intact?

Or perhaps this is a duplicate of #273 ?

afriza, 2018-09-04:
I'm not sure if the image is supposed to behave that way, but even after:
- initial flashing of image
- erasing/formatting the sdcard using Disk Utility (under macOS)
- flashing the image again
The passwords and configuration are still there.

Therefore I don't think it is duplicate of #273 since #273 erases Master Boot Record only.

lurch, 2018-09-04:
I don't have a Mac myself, but does Disk Utility have a "full format" / "full erase" option?

afriza, 2018-09-04:
I don't see any option related to "full erase/format" during erase

[image]

What is supposed to be different when using "full format/erase" ?

lurch, 2018-09-04:
A "quick format" (which is what it appears to be doing) will only overwrite the metadata at the start of the disk, whereas a "full format" would overwrite the entire contents of the disk (and hence be much slower).

I just did a quick search and found https://www.howtogeek.com/179284/how-to-securely-wipe-a-hard-drive-on-your-mac/

afriza, 2018-09-06:
Hi @lurch, thanks for the link. However it seems that the `Security Options...` is [not](https://discussions.apple.com/thread/7317599) [available](https://discussions.apple.com/thread/7258264) for flash-based storage including sdcard.

jhermsmeier, 2018-09-06:
This is odd – reflashing the same image that doesn't contain any config should still reset the filesystem's allocation tables, as well as overwrite the particular config files, and those values shouldn't be there anymore.

> then set password and do some configurations

@afriza can you describe which configurations etc. exactly you've set there? I'd like to see if I can reproduce this.

lurch, 2018-09-06:
@jhermsmeier as the image is labelled "sysupgrade" I was guessing that maybe it's small enough to only overwrite the boot / OS partitions at the start of the disk, and leaves a settings partition towards the end of the disk untouched? (which would explain the behaviour described here).
But that's a _total_ guess so I might be _totally_ wrong :laughing:

jhermsmeier, 2018-09-07:
I've had a look at the images; the sysupgrade image has a FAT16 & squashfs partition, the others seems to have two squashfs partitions. Since squashfs is read-only by design (unless unsquashed & resquashed), I suspect there's something else going on here, but I'd need more info about the process steps, devices and all to be able to tell.

lurch, 2018-09-07:
AFAIK the Raspberry Pis are only able to load their firmware (`bootcode.bin`, `start.elf` etc.) from a FAT16 / FAT32 partition, so a disk image that _only_ has squashfs partitions indeed sounds very strange :man_shrugging:

closed 2018-10-15 by jviotti