balena-io/etcher #2468: The `driveBlacklist` setting is being compared against drives `devicePath`.

zvin, 2018-09-05:
The `driveBlacklist` setting is being compared against drives `devicePath`.
It should be compared against `device` or `raw` as not all devices will have a `devicePath` (internal mmcs for example).
We also need to add a setting to blacklist by `devicePath`.

Here: https://github.com/resin-io/etcher/blob/master/lib/gui/app/app.js#L245

lurch, 2018-09-06:
Related to https://github.com/resin-io-modules/drivelist/pull/297 and / or https://github.com/resin-io-modules/drivelist/pull/302 ? I wonder if it might be easier to add dummy devicePaths in drivelist?

zvin, 2018-09-06:
@lurch the 2 PRs you mentionned are merged.
Why would we add "dummy" devicePaths?
The `devicePath` is the name of the device in `/dev/disks/by-path/` (other places on win32 and mac), its purpose is to know where the device is plugged (for example to blacklist some usb  ports).
 * `device` is `/dev/sda`
 * `devicePath` is `pci-0000:00:14.0-usb-0:1.1.3:1.0-scsi-0:0:0:0` (on Linux)

lurch, 2018-09-06:
Sorry, feel free to ignore me - it's so long since I've looked at this that I've forgotten how most of it works. I just thought it _might_ be easier to add dummy devicePaths in drivelist for those devices that don't have them (e.g. `internal-mmc-slot`) than it would be to check against two separate blacklists in Etcher :man_shrugging:

closed 2018-10-15 by jviotti