balena-io/etcher #2471: command diskutil list -plist returned without data

varunkalyan, 2018-09-08:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

varunkalyan, 2018-09-08:
[image]

varunkalyan, 2018-09-08:
Unable to flash a USB and see the above error

thundron, 2019-01-26:
Which balenaEtcher version are you using @varunkalyan @sidouglas ? Did you try that command by itself to check its output? `diskutil list -plist`

thundron, 2019-01-26:
Also, when are you seeing this error? During image selection, drive selection or when clicking the flash button? Also please fill in the information above on the issue template

zvin, 2019-04-03:
Etcher no longer uses diskutil since v1.5.0

closed 2019-04-03 by zvin