balena-io/etcher #2473: Always prompted to upgrade to 1.4.3 (1.4.4, macOS)

dowlingw, 2018-09-10:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** macOS 10.13.6 (17G65) on MacBook Pro (Retina, 13-inch, Mid 2014)
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** No

Etcher 1.4.4 on macOS appears to be prompting the user to update to `1.4.3` every time the application launches.

The bottom-right-hand corner of the application shows `1.0.0`, so suspect the build process may not be setting the application version information correctly.

Is there a cli flag to start Etcher's electron instance with developer tools open _on load_? This way I could capture the outbound upgrade check request for this ticket.

jhermsmeier, 2018-09-10:
@dowlingw Where did you download / install Etcher from, in what kind of package (dmg, or zipped app, etc.), and when?

dowlingw, 2018-09-11:
Ugh, sorry buddy - looks like spotlight was finding a random older copy somewhere else.

Thanks again for the awesome tool!

closed 2018-09-11 by dowlingw