balena-io/etcher #2474: Image Incorrectly Verified

ptamike, 2018-09-10:
Etcher version: 1.44
Operating system and architecture: Raspbian PIXEL desktop (Stretch) on Gigabyte BG-BXBT-1900 64-bit Intel PC
Image flashed: The image is a standard format Raspberry Pi image with Raspbian Stretch

Problem: Etcher reports successful verification even though the target and source images are different!

I  copy of lots of cards for Raspberry Pi SBCs using the command-line tool dc3dd, but would like to change to using Etcher. However, I've been experiencing a high proportion of failures. Having spent some time investigating this, I think I have spotted a problem. 

The Pi image is being flashed to Kingston 8GB class 4 microSD cards using a Lexar Workflow UR1 3-slot microSD reader/writer.
If the target microSD card is unformatted and showing as all free space (using Disks), Etcher will run as expected, verify the card and report success. However subsequent checking of the card shows that the intended image has not been copied but the target micoSD card is left with an empty FAT32 partition and 4.2MB of free space. It should have written the standard Pi Boot and Ex4 Partitions. Incidentally, if the target microSD card is subsequently plugged into a Windows PC it is unreadable and needs to be repaired and formatted before it can be used again.

The serious failing for me is the fact that Etcher can report successful verification when the target image clearly does not match the source!

Any suggestions would be very welcome.

Mike

lurch, 2018-09-11:
That indeed sounds very strange! :confused: 
Posting the full devtools log might help the Etcher developers to investigate...
I wonder if it might be related to #1523 ?

jhermsmeier, 2018-09-11:
Do I understand it correctly that you're on Windows?

ptamike, 2018-09-11:
Jonas,

No. As I stated at the top of the issue, the problem is occurring on a Gigabyte Intel mini PC running Raspbian PIXEL desktop.

Regards,

Mike

jhermsmeier, 2018-09-11:
Oh, sorry I mistook that for the image being flashed. How did you install Etcher (deb package, AppImage, etc)?

closed 2018-10-15 by jviotti