balena-io/etcher #2476: Etcher commit guidlines does not match resin commit requirements

amdomanska, 2018-09-12:
Etcher commit guidlines (https://github.com/resin-io/etcher/blob/master/docs/COMMIT-GUIDELINES.md) require subject of the commit to NOT have the first letter capitalized. On the other hand resin commit lint (https://github.com/resin-io/resin-commit-lint) requires that to be capitalized. That way someone who follows the etcher guidlines end up with filed commit check after submitting PR

lurch, 2018-09-12:
@jviotti will know more, but IIRC the Etcher versionist config is also deliberately different to the "standard" versionist config.

jhermsmeier, 2018-09-12:
@lurch the resin-wide commit format has changed a bit, but Etcher's commit guidelines haven't caught up with it yet ;)

lurch, 2018-09-12:
/me crawls back under his rock :wink: