balena-io/etcher #2490: No removable drive detected

crystaldavidson, 2018-09-30:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** mac os x yosemite (10.10.5) and macbook air (early 2015)
- **Image flashed:** motioneyeos-raspberrypi3-20170212.img
- **Do you see any meaningful error information in the DevTools?** unknown

Hello - I am trying to build a security camera with my raspberry pi 3. I just need to get the motion eye os onto my pi but etcher unfortunately won't detect my removable drive (microSD). I found an older post that mentioned their sd card not registering and they believed their macbook was too old to be compatible with etcher 1.4.4. 

Please help as I am near the end of the completion of my security camera project and am very excited to catch neighborhood rascals.

kriekprinsloo, 2018-09-30:
My MacBook Pro is late 2012, and has no problem with writing cards, so it
probably not that.  Try opening the terminal and finding all the volumes on
your computer:

The command is diskutil list

Upon entering the command, you should find a list similar to mine:

/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE
 IDENTIFIER
   0:      GUID_partition_scheme                        *500.1 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Macintosh HD            250.4 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3
   4:                  Apple_HFS Linux                   200.0 GB   disk0s4
   5:                  Apple_HFS Untitled                48.6 GB    disk0s5

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE
 IDENTIFIER
   0:                  Apple_HFS Macintosh HD           +250.0 GB   disk1
                                 Logical Volume on disk0s2
                                 29F46527-89DE-41C9-AD23-12EC1307C044
                                 Unencrypted

Try this command and see if you can find your SD card, then get back to me
with the results.

crystaldavidson, 2018-09-30:
thank you so much! 

my SD card came up "CRYSTALCAN"

Result of  command is diskutil list:

/dev/disk0
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage                         250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3
/dev/disk1
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Macintosh HD           *249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 45A7CCB9-8249-4762-BBD1-AC5F4A56EA09
                                 Unlocked Encrypted
/dev/disk2
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *15.9 GB    disk2
   1:             Windows_FAT_32 CRYSTALCAN              15.9 GB    disk2s1
Crystals-MacBook-Air:~ RadicalSpirit$

kriekprinsloo, 2018-10-01:
Ok, this is great information.  We can now isolate that the problem is most
probably Etcher's fault, as your computer sees the drive.  The next step I
can recommend is to try and replicate the error, and see if the DevTools
gives you any error messages to work with.

To open the DevTools you need to hit Command+Alt+i, and it should pop right
open.  See if then the console spits out an error message, and we'll work
from there.
Thanks!

crystaldavidson, 2018-10-01:
Not sure if this is an error message or not.

Copy and pasted the console text below:

etcher:sdk:scanner scanner:init +0ms Scanneradapters: Map(2)domain: nullisScanning: trueonDevices: function ()onError: function ()options: Object_events: EventHandlers_eventsCount: 2_maxListeners: undefined__proto__: EventEmitter
gui.js:105081 etcher:sdk:scanner subscribe +5ms BlockDeviceAdapterdevices: Array(0)domain: nullid: "blockdevice"_events: EventHandlers_eventsCount: 2_maxListeners: undefined__proto__: EventEmitter
gui.js:105081 etcher:sdk:scanner subscribe +2ms USBBootAdapterblobCache: Objectdevices: Array(0)domain: nullid: "usbboot"progress: Object_events: EventHandlers_eventsCount: 2_maxListeners: undefined__proto__: EventEmitter
gui.js:105081  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.4.4, Type = dmg
gui.js:105081 Sun Sep 30 2018 18:32:12 GMT-0700 (PDT) Application start ({"packageType":"dmg","version":"1.4.4"})
gui.js:105081 Sun Sep 30 2018 18:32:12 GMT-0700 (PDT) Checking for updates ({"currentVersion":"1.4.4","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
gui.js:105081 etcher:sdk:scanner start +154ms true
gui.js:105081 Sun Sep 30 2018 18:32:13 GMT-0700 (PDT) [object Event]
gui.js:105081 Sun Sep 30 2018 18:32:15 GMT-0700 (PDT) Update notification skipped ({"reason":"Latest version"})
gui.js:105081 Sun Sep 30 2018 18:37:37 GMT-0700 (PDT) Open image selector
gui.js:105081 Sun Sep 30 2018 18:40:21 GMT-0700 (PDT) Select image ({"path":"/Users/RadicalSpirit/Downloads/motioneyeos-raspberrypi3-20180224 (1).img","extension":"img","size":{"original":230686720,"final":{"estimation":false,"value":230686720}},"hasMBR":true,"hasGPT":false,"partitions":[{"type":14,"id":null,"name":null,"firstLBA":2048,"lastLBA":43008,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":43008,"lastLBA":411648,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
gui.js:105081 Sun Sep 30 2018 18:40:23 GMT-0700 (PDT) Open modal ({"name":"drive-selector"})
gui.js:105081 Sun Sep 30 2018 18:41:14 GMT-0700 (PDT) Modal accepted ({"name":"drive-selector"})
gui.js:105081 Sun Sep 30 2018 18:41:23 GMT-0700 (PDT) Open modal ({"name":"drive-selector"})
gui.js:105081 Sun Sep 30 2018 18:41:26 GMT-0700 (PDT) Modal rejected ({"name":"drive-selector","method":"backdrop click"})
gui.js:105081 Sun Sep 30 2018 18:42:10 GMT-0700 (PDT) Open modal ({"name":"drive-selector"})

forsakensingularity, 2018-10-01:
I used Etcher yesterday and it worked fine BUT today it does not see my DVD drive; other apps see the drive fine but Etcher cannot; so I downloaded the CLI and ran --

sudo /opt/etcher-cli/etcher -d /dev/sr0 -y -c -u haiku-r1beta1-x86_64-anyboot.zip 

And it came back after several seconds and said -- 

The selected drive "/dev/sr0" was not found

We can't find "/dev/sr0" in your system. Did you unplug the drive?

Also, Command-Alt-i does not work on my Linux Mint 19 system.  I have an AMD Athlon processor with 4GB RAM ... like I said, it was just working but has stopped -- weird...

kriekprinsloo, 2018-10-01:
@crystaldavidson I see...

The DevTools didn't show any error I could spot.  Does Etcher see your other drives like your laptop's hard drive?  I would also try to test a different SD card.  If these suggestions do not work to your liking there's one other idea I had.

@forsakensingularity The DevTools are accessible by the shortcut `Ctrl+Shift+I`, though if you are using the CLI I think that that is the error message.  You said that you were trying to write a DVD, what kind is it?  The main types are DVD-RW and DVD-R.  DVD-RW (read-write) are like SD cards and flash drives.  You can write them over and over again, while DVD-R (read) are only able to be written once.
Also, are you using an external CD/DVD drive or a built-in one?

Thanks!

forsakensingularity, 2018-10-02:
DVD-RW... target does not matter as it does not see the internal drive.

I'll play with bios settings...

crystaldavidson, 2018-10-02:
Reinstalled Etcher and laptop hard drive now showing up so that's great! Tried a different SD card but nothing shows up still.

stuzog, 2018-10-16:
I'm having the same problem on Trying to burn linuxmint-19-cinnamon-64bit-v2-1.iso in El Capitan 10.11.6 on an iMac 7,1 6GB RAM. Etcher 1.4.5 won't recognise a formatted 32GB SanDisk Cruzer USB stick, even though it appears on the desktop. "No removable drive detected." Etcher used to work -- what has changed?

forsakensingularity, 2018-10-16:
I personally have no idea.  It will burn to a SanDisk drive; I even got it
to write to 5 at once but it no longer sees DVDs - so I have stopped using
Etcher.

Writing to USBs are fine but for my purposes I need an expandable medium so
writing a DVD and installing is a longer process but it works. usb drives
are not great if you write the OS but you can't expand it to the whole disk
- like is possible with OpenSuSE ...  live and learn.

crystaldavidson, 2018-10-17:
hopefully we can fix this bug. 

i have been trying my best to work in terminal so I can  skip the entire etcher process, which is a shame because etcher has a beautiful user interface and is highly recommended for raspberry pi projects.

fjblau, 2018-12-22:
Etcher no longer recognizes an SD card as a removable drive. Kind of renders the program useless. YOU HAD ONE JOB, as the kids say.

thundron, 2018-12-31:
@fjblau @crystaldavidson Did you try @stuzog 's steps by chance?
Also, if you want to flash system drives (which is **highly** suggested not to do so), you can enable `unsafe mode` in Etcher settings

lurch, 2018-12-31:
@forsakensingularity Etcher doesn't support burning DVDs, see #2432

thundron, 2018-12-31:
Closing as duplicate of https://github.com/balena-io/etcher/issues/2086

RobinBrackez, 2019-01-01:
I had the same problem on Linux Mint. My micro sd card couldn't be found by Etcher, but was visible in the file manager. I enabled the "unsafe mode" in the settings of Etcher and then the sd card showed up. The sd card is 64GB which is "unusually large" for an sd card according to Etcher (got a warning about it), so it probably thought the drive wasn't suitable to flash.

sartor, 2019-01-05:
I have same problem with 1.4.x version on LinuxMint But version 1.3.1 work perfect!

thundron, 2019-01-07:
@sartor Did you try enabling `unsafe mode` in the settings?

sartor, 2019-01-08:
Yes. No changes

whatisc, 2019-02-11:
I am having this issue too, unsafe mode leads to no changes.

jasalazar, 2020-06-16:
I have the same issue. I enabled the `unsafe mode` and it detected the micro sd, but etcher won't start the process after clicking "Flash!"

anthonym01, 2020-06-16:
Enable unsafe mode, and make sure your os is detecting the drive

closed 2020-09-28 by thundron