balena-io/etcher #2497: Etcher 1.4.4 Ignores Privacy Setting

mojoTX, 2018-10-06:
- **Etcher version:** 1.4.4
- **Operating system and architecture:**  macOS 10.14 "Mojave" and MacBook Pro (Late 2016)
- **Image flashed:** ubuntu-18.04.1-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?**  No

I have the Little Snitch firewall installed, and when I installed Etcher and tried using it I was surprised at the number of outbound connections it made. 

I'm posting an issue because if you go into the settings and de-select the option, "Anonymously report errors and usage information to resin.io," the app ignores the setting and still makes a large number of connections to well-known web usage sites such as Mixpanel, GoSquared, and Google.

I do not know if this is a bug in the Etcher code, or if it's cruft that is baked into the Electron JS framework. Regardless, it's misleading to give a user the option to opt-out of usage reporting, and still make the connections anyway. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

thundron, 2018-10-16:
This is due to the services being registered but not really used with the flag disabled, so while it won't cause any harm, we can surely prevent the registration in the analytics module.
For instance, at this line here https://github.com/resin-io/etcher/blob/master/lib/gui/app/modules/analytics.js#L33 you can see what I mean :)

lurch, 2018-10-16:
I wonder if some of those warnings are triggered by a) the Etcher auto-update check and b) the Etcher success-banner?

thundron, 2018-10-16:
@lurch I was looking into the latter as well, but good guess on the auto-updater. It should be the call to AWS S3

SilverWolf32, 2018-10-29:
@thundron If it's making the network connections regardless, how is that not harmful? Even if it doesn't send any _data_, it still tells Google and the like that you're using Etcher.

LeBaux, 2018-11-10:
Why bootable USB maker needs so much analytical data and [connection to Facebook](https://forums.balena.io/t/serious-privacy-concerns-with-etcher-1-4-4/4103)? Ignoring the user preference just takes the cake, please address this issue. Thank you.

kevcampb, 2018-12-05:
There is also issue #2057 

It appears that ether defaults to sending errors and usage data, and the program is opt-out rather than opt-in. Certainly it's not GDPR compliant if it is.

SilverWolf32, 2019-01-28:
Checking back in, this is still a thing: the latest v1.4.9 is still using network with analytics turned off.

SilverWolf32, 2019-01-28:
Also, I can confirm that this is **not caused by Electron**. My own tiny Electron app is showing 0 bytes of network usage.

SilverWolf32, 2019-01-28:
I'm trying to disable analytics in a local build, including replacing all of [analytics.js](/balena-io/etcher/blob/master/lib/gui/app/modules/analytics.js) with a stub, and it's _still_ making network calls to tracking services. How do I properly disable that?

thundron, 2019-01-29:
@SilverWolf32 We're working on it, it's because we have a webview in the final flashing page that loads up analytics too and we need to pass the opt-out flag when connecting to that url too (we found out we currently don't), can't tell you when it'll be available though as we're making some major updates related to electron & npm first.
This is without doubt not an issue in electron itself so you're not wrong on that, just wanted to let everyone know that it's on our to-do list

sneak, 2019-03-29:
[image]

lol

SilverWolf32, 2019-03-29:
There's _ads_ in the software now? That's kind of annoying. Good thing it's open-source, and easy to get up and running.

sneak, 2019-03-29:
look at what the ad is for - setting up a raspberry pi to block the kind of trackers etcher ships turned on by default. definitely a “do as i say, not as i do” kind of thing.

unikitty37, 2019-04-28:
Since the software is not compliant with the law as it stands, could a release be made that removes tha analytics *completely*?

They can then be added in once they're strictly opt-in, which is what the law requires.

sneak, 2019-04-28:
Can we ask the fundamental question of “what possible benefit does an application whose function is ‘flash file a to block device b’ derive from networked real-time surveillance?”

Like, what is the best case scenario for the benefit of receiving analytics from the user base? What is this accomplishing? Why is this even a discussion?

mojoTX, 2019-04-29:
I completely agree with @sneak. This is a graphical front end for a simple process: flashing a file to a block device. It replaces having to run something like `dd if=file.img of=/dev/disk2 bs=512`. 

Analytics might be interesting, but they aren't essential. I think the problem is that this project did not arise as an open-source idea from an individual committed to OSS principles--instead, it was released by Balena, a start-up software company, and I suspect THEY are the ones who want to leverage this tool for commercial purposes. I note that they are working on an "[Etcher Pro](https://www.balena.io/etcher/pro)" product.

Maybe it's time for someone to fork this project and rip out the spyware.

SilverWolf32, 2019-04-29:
To be fair, company != bad. And I think their Etcher Pro is actually a hardware device. But yes, forking does sound like a good idea.

mojoTX, 2019-04-29:
I understand that, @SilverWolf32. Companies are a GOOD thing, because developers have to eat!  The money to live has to come from somewhere, and donations to open-source projects are seldom sufficient for developers to survive, unless it's a really popular project, AND the developers are willing to live frugally! 🙂

My point is that corporations DO love analytics and marketing. I've worked for some companies where product management was a separate department, and their job was to dream up new ways to squeeze more revenue from our customers. I do not know if the analytics were inserted at the request of the Balena upper management, or if someone just thought it would be nice to have. It doesn't really matter. It needs to be fixed, and eventually it probably WILL be fixed, but it doesn't sound like it is a big priority to the core team of developers, at least, not compared to other issues.

In theory, there is nothing stopping someone from forking the code and removing the analytics. However, it only takes a few minutes to complain on Github, but it would take hours to fork the project and fix it correctly. That's why we complain, but aren't fixing it.  I am guilty of that myself!  😄

I think the app is a very good app, and I am grateful for the efforts of the people who created it. If I can ever find the time, I would be willing to see if I could fix it myself. However, I stay pretty busy with my own job, as well as spending time with my family, etc. Since I didn't pay anything for the software, I cannot complain too much!

thundron, 2019-05-07:
I understand the doubts you might have about sending out data to what is in fact a company you don't know everything about, but *our* analytics are anonymous and as such they're only used to monitor various things about the application itself, mainly monitoring how good/bad each release is doing in terms of usability and old/new errors that we try to solve.
If we were to base our development process on user feedback *only* then the application itself would suffer, while our goal is and has always been to provide the best experience with Etcher as well as keeping it real so that a couple users saying that Etcher is bad doesn't affect the other hundreds of thousands users that use it everyday and I'm not talking about you of course, it's to put it into perspective as to why anonymous analytics aren't such a bad thing even in a simple UI layer for flashing like Etcher is (as of now, at least).

I also understand this isn't the first thing that comes to mind for sure when seeing that deactivating analytics *still* doesn't stop a couple calls, which is why github issues are a thing and I thank you for being active and having opened this issue specifically! So returning to the subject of this issue, we recently updated our code so that external content also reads the opt-out anonymous analytics config and behaves accordingly (e.g. the webviews we have for the featured project and the success banner which we revamped recently and needed some adjustments).

Closing this one, feel free to open a new issue if it happens again. It's what we're here for!

closed 2019-05-07 by thundron

unikitty37, 2019-05-07:
You misunderstand.

If Etcher is used by an EU citizen, the law *requires* that you not collect any analytics until that user has given explicit, informed, OPT-IN consent.

It’s also unlawful to tie that consent to other things — so you can’t disable functionality until the user consents.

I suggest you look at the maximum penalties for GDPR infringements and feed back to management exactly what liabilities these analytics are exposing you to — they might be a bit less keen on them at that point.

SilverWolf32, 2019-05-07:
I completely agree that analytics should definitely be opt-in (although disabling them properly is essential even without that, thank you @thundron).

But @unikitty37, you mention that consent can't be tied to other things. Does that apply here? I don't recall disabling analytics ever disabling some feature of Etcher along with it. Although I haven't used Etcher in a while, so I suppose they could have added that in recently.

unikitty37, 2019-05-07:
I wasn’t claiming that Etcher was tying consents so much as mentioning it as part of the GDPR requirements — apologies for being unclear.

(Some companies have been known to add that sort of requirement when they’re told they need to get consent first, so I thought it was worth adding up front:)

sneak, 2019-05-11:
> but our analytics are anonymous and as such they're only used to monitor various things about the application itself, mainly monitoring how good/bad each release is doing in terms of usability and old/new errors that we try to solve.

it turns out that you don’t get to crowdsource your functional testing without the consent of those whose computers and electricity you are using to run those mass functional tests.

LeBaux, 2019-07-02:
I see this issue is marked as closed, yet https://github.com/balena-io/etcher/blob/master/CHANGELOG.md does not mention anything about addressing the problem. How was the privacy issue fixed?

Note that I generally do not have problem opting-in if I am asked nicely and the purpose is explained. But opt-in by default is the reason I left Windows.

Thank you for clarification.

thundron, 2019-07-05:
@LeBaux We have an ongoing internal discussion on the changes to perform in order to have a better UX on the analytics side.
This issue though was intended for a bug: when deactivating the analytics, *some* were still being sent, which wasn't intended and has since been fixed (and this issue closed)

LeBaux, 2019-07-05:
Thanks for clarification. I hope Etcher will continue to thrive while meeting expectations of sometimes demanding & grumpy Linux userbase.

thundron, 2019-07-05:
We're doing our best to keep everyone happy and find a middle-ground where all can stay :P

sneak, 2019-07-05:
The only demand that I have is that the software not do things to my computer without my consent. That’s the difference between a tool and malware.

LeBaux, 2019-07-06:
@sneak next time maybe start off with reading the comments (more than last 2) before bringing your staff of rushed ethical judgement. Cheers!

sneak, 2019-07-06:
https://github.com/balena-io/etcher/issues/1718

2017 to 2019 (problem still exists!) is not “rushed”.

sneak, 2019-07-06:
https://github.com/balena-io/etcher/issues/2057

sneak, 2019-07-06:
Etcher still has this bug; information is transmitted without the user’s consent.

SilverWolf32, 2019-07-07:
Well, to be fair, I would argue that that's a _different_ bug; while it was certainly mentioned in this thread, the issue was originally about the opt-out checkbox not working.

We should definitely carry on with the other issue, though!

saitam1, 2019-07-19:
just saying http://www.enforcementtracker.com/

rradar, 2019-08-25:
Actually this could turn out really bad. Balena Inc know that they have gdpr violations since august 2017 (https://github.com/balena-io/etcher/issues/1718). Regarding the fine this could be a reason to quote a big amount and not just 20k or so for educational propose :grimacing:

Maybe the legal team of balena (legal@balena.io) isn't aware of this? @alexandrosm is?

muscarina, 2019-11-03:
Has anyone made a GDPR violation report on this yet? If not, it's about time.

bztsrc, 2020-02-21:
FYI, there's an alternative that does not collect analytics on you, so it is 100% GDPR compliant. https://gitlab.com/bztsrc/usbimager

It can read compressed images, able to verify writes, has multilingual support, and shockingly no more than 128 kilobytes. FOSS and MIT licensed, give it a try if you'd like.

Cheers,
bzt

sneak, 2020-02-21:
That‘a great! Thank you for the link. 

I am also considering just forking Etcher, removing the trademarks and spyware, and re-releasing.

rradar, 2020-03-07:
Ok. Seriously?

- **etcher** size: >400MB (>100MB app + ~300MB dependencies)
- [**usbimager**](https://gitlab.com/bztsrc/usbimager) size: **0**,25MB

Essentially they do the same: _uncompress, write, verify_

Can it really be that etcher uses almost half a gigabyte more of code just fore _advanced features_ like spyware, adware (or modern 'promoted content') and security risks?

bztsrc, 2020-03-11:
> Ok. Seriously?

Seriously. :-) I've added more translations to it, so USBImager is now 251 kilobytes uncompressed. It is a single portable executable, no additional files nor dependencies, and it is not expected to grow much bigger (maybe a little bit if more translations added in the future).

Cheers,
bzt

rradar, 2020-03-18:
`apt purge balena-etcher-electron `
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  balena-etcher-electron*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 429 MB disk space will be freed.
```
429MB :free: :eyes:

muscarina, 2020-03-21:
You say bzt? I say MVP.

gafarma89, 2020-05-23:
@bztsrc thank you for your privacy-respected fork.

sinni800, 2020-05-24:
It's not a fork, it's a different implementation...

But that a USB IMAGE WRITER has privacy considerations is just funny. Sad.

thundron, 2020-05-29:
What's also sad is watching people not understand the difference between personal and anonymous data.
I think the discussion is long concluded and there's no need to further comment, we've been extremely clear in both admitting what was (previously) an issue and clarifying what our scope is after fixing those.