balena-io/etcher #2498: As a weird user, I need to pick the drive myself, so that I don't erase a 1.6 TB drive by accident, ever again :)

alicerocheman, 2018-10-07:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Windows 7 Pro

Hey,

Thank you for your work, I love Etcher!

I would love to have an option in the settings, to uncheck autoselect for the drive.

Given that it will erase it, I'd like to be sure I pick it myself every time, so I'd rather have the drive unselected by default.

I'd understand if it's too much trouble though!

Thanks for being cool :)

kriekprinsloo, 2018-11-19:
Yes, sorry for my late reply.  The external hard drive is just as valid for writing as a thumb drive or SD card.  These things do happen sometimes.  One way to help you to better determine which drive is the correct one would be to use command line tools to scope out which device is the correct one.

You want to open command prompt and use:
`diskpart`

After this you will enter the diskpart software.  Here you should use the command
`list disk`

It should help you determine the right disks in the future

louy2, 2018-12-18:
What I'd like to suggest is that the burning process is a destructive process after all, so confirmation popups may be helpful in making sure the users don't lose data over a simple mishap. Adopting the heuristic that any disk over 1TB is probably an error, another confirmation popup may help reduce error even further. 

I'd gladly submit a PR if someone may guide me to where key related components are located.

thundron, 2018-12-31:
This should be fixed in latest versions (I couldn't reproduce anymore). Closing

closed 2018-12-31 by thundron