balena-io/etcher #2502: Make Etcher detect usb boot devices without running with sudo

cyplo, 2018-10-10:
- **Etcher version:** etcher-electron-1.4.4+7e01eca-x86_64.AppImage
- **Operating system and architecture:** Fedora 28, x86_64 - Gnome3 under Wayland
- **Image flashed:** any
- **Do you see any meaningful error information in the DevTools?** no - does not start

Background:
Currently, to be able to flash USB boot device, like Balena Fin, Etcher needs elevated privileges when started to be able to detect such a device. This is currently achieved with running etcher via `sudo`.
 
My system runs Gnome3 under Wayland, this is what Fedora 28 does by default. 
Under Wayland, applications cannot be ran with sudo by default - see this bugzilla [issue](https://bugzilla.redhat.com/show_bug.cgi?id=1274451).
When ran, they produce error like so:
```
sudo ./etcher-electron-1.4.4+7e01eca-x86_64.AppImage
[sudo] password for cyryl: 
No protocol specified
```
Currently Etcher can ask for elevated privileges already, when ran normally, without sudo, when trying to write to the disk. This behavior works under Wayland.

Could we make sure that Etcher can be ran without sudo and then ask for elevated privileges for the device detection phase as well ? This would help any Wayland users, which there will be more of, as distros move away from X server, to Wayland land.

p.s. if you found this issue and looking for an immediate fix for Fedora - click on the gear icon on the login screen and choose `Gnome on X` instead of `Gnome` - this should launch Gnome without Wayland, thus allowing for running Etcher GUI with `sudo`

thanks !

lurch, 2018-10-19:
See also #1819

Roesjka, 2019-01-09:
https://unix.stackexchange.com/questions/422040/will-wayland-ever-support-graphical-sudo

also references the wayland issue for other applications and gives 2 solutions.
Might be off some help.