balena-io/etcher #2508: Expected a number

freekvh, 2018-10-16:
- **etcher-electron-1.4.5-x86_64.AppImage** 
- *KDE Neon 18.04:** 
- **none** 
- **Do you see any meaningful error information in the DevTools?** 

Etcher is unresponsive and can't be used.

The Error:

TypeError: Expected a number
    at prettyBytes (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/pretty-bytes/pretty-bytes.js:18:10)
    at module.exports.getDrivesSubtitle (file:///tmp/.mount_OSOdsn/usr/bin/resources/app.asar/generated/gui.js:7477:14)
    at fn (eval at compile (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:15351:15), <anonymous>:4:182)
    at expressionInputWatch (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:16352:31)
    at Scope.$digest (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:18002:34)
    at Scope.$apply (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:18280:24)
    at /tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:20127:36
    at completeOutstandingRequest (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:6274:10)
    at /tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/angular/angular.js:6554:7
    at wrapped (/tmp/.mount_OSOdsn/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:444:21)

freekvh, 2018-10-18:
It could have something to do with opening Etcher twice very quickly after each other (KDE does not require double clicks for opening... took some getting used to). It did work normally the last 2 times I tried.

Jacalz, 2018-10-21:
I am getting the exact same error every time I start Etcher on Solus after upgrading to the latest version, see picture below:
[image]

Kuoi, 2018-10-23:
I did get this too , but rightclicked on the file , > preferences > set all rights to 'read and write' and enabled "make it run as a program" (translated from dutch :-) ) 
Then on opening i answered "no" on the question for the desktop file , and it opened without the warning.
I don't know what did the actual trick to let it open fine ,but this way it works like it should

lukasmedia, 2018-11-02:
Same on Ubuntu 16:

[image]

thundron, 2018-11-05:
Seems to be related to read/write permissions given to etcher *before* running it.
Did you all try this procedure https://github.com/balena-io/etcher/issues/2508#issuecomment-432415759 ?

ivangsm, 2018-11-06:
The same thing happened to me after I installed the `.desktop` file when I ran the `.appimage` file. The way I fixed it was to delete the file `appimagekit-Etcher.desktop` which is in the path:  `~/.local/share/applications`

matejmosko, 2018-11-06:
> Seems to be related to read/write permissions given to etcher _before_ running it.
> Did you all try this procedure [#2508 (comment)](https://github.com/balena-io/etcher/issues/2508#issuecomment-432415759) ?

Doesn't work for me.

thundron, 2018-11-07:
We chased down the issue and it seems to go all the way back to nodejs itself, which has been fixed in new nodejs versions.
We're working on updating electron used by etcher so we'll include the newer nodejs versions and fixes

Jacalz, 2018-11-07:
I was using the Solus version and the thing is that they are building with NodeJS version 8.  In what version can the bug be found? Or will this simple get fixed with electron getting updated? @thundron

dragon788, 2018-11-14:
I only ran into this issue when attempting to "launch/open" the AppImage via the quicksearch `Super` menu in Gnome (Ubuntu 18.04). If I instead opened "Files" and ran the AppImage from Nautilus (Gnome file explorer) it ran correctly.

Xosmond, 2018-11-15:
I just chaged the permissions on the Iso file and the application downloaded to 777 and it worked, i think you can debug with the permissions and get the problem.

closed 2019-05-09 by zvin