balena-io/etcher #2511: USB Not bootable after Succesfull Flash via Etcher

wulfalpha, 2018-10-18:
- **Etcher version:** 
1.4.5
- **Operating system and architecture:** 
Mac OS X
- **Image flashed:** 
Manjaro Budgie
- **Do you see any meaningful error information in the DevTools?** 
No

As far as I can tell the Iso is good as it is the latest. I've used 1.4.4 to successfully flash a drive for Manjaro Gnome once. I've tried two different USB drives on the same computer. It just won't boot from the USB. I'm thinking the problem may be that the computer is old, and it might not see the drive as bootable. I'm giving it another Go with UNetbootin to try and see if the problem is with etcher or my computer.

thundron, 2018-10-18:
Hi there,
We don't apply any additional logic beside flashing, so for things like bootable USB cards you'll need a different tool.
More on that here https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

closed 2018-10-18 by thundron

thundron, 2018-10-18:
Sorry man, I just noticed you said you flashed a Manjaro Budgie image, which might fall into [this](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos) section then.
Let us know how it goes with UNetbootin, but without further information I think there's not much we can do about it