balena-io/etcher #2517: Request password privileges

KyKyIIIKuH, 2018-10-24:
- **Etcher version:** 1.4.4
- **Operating system and architecture:**  Debian 9.5 x86_64 Linux 4.9.0-8-amd64
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 
- KDE Plasma 5.8.6

A window appears asking for the password.
I tried to enter the root password and user password.
This password box has reappeared.

[image]

thundron, 2018-10-31:
Did you try it with the latest version too? This was due to a package trying to use sudo and kdesudo at the same time iirc, which should be fixed in 1.4.5 and 1.4.6

KyKyIIIKuH, 2018-11-01:
1.4.6 same problem