balena-io/etcher #2518: RangeError: toString() radix argument must be between 2 and 36

fabianMendez, 2018-10-24:
- **Etcher version:** Etcher-Portable-1.4.5-x86
- **Operating system and architecture:** Windows 7 32 bits
- **Image flashed:** [ubuntu-18.10-desktop-amd64](https://www.ubuntu.com/download/desktop/thank-you/?version=18.10&architecture=amd64)
- **Do you see any meaningful error information in the DevTools?** 
RangeError: toString() radix argument must be between 2 and 36
    at Number.toString (<anonymous>)
    at Stream.hash.once (C:\Users\<user>\AppData\Local\Temp\nsv6928.tmp\app\resources\app.asar\lib\sdk\writer\checksum-stream.js:103:45)
    at Object.onceWrapper (events.js:293:19)
    at emitNone (events.js:86:13)
    at Stream.emit (events.js:188:7)
    at emitReadable_ (_stream_readable.js:434:10)
    at emitReadable (_stream_readable.js:428:7)
    at readableAddChunk (_stream_readable.js:189:13)
    at Stream.Readable.push (_stream_readable.js:136:10)
    at Stream.Transform.push (_stream_transform.js:128:32)
    at Stream._flush (C:\Users\<user>\AppData\Local\Temp\nsv6928.tmp\app\resources\app.asar\node_modules\xxhash\lib\xxhash.js:47:10)
    at Stream.<anonymous> (_stream_transform.js:118:12)
    at Object.onceWrapper (events.js:293:19)
    at emitNone (events.js:86:13)
    at Stream.emit (events.js:188:7)
    at prefinish (_stream_writable.js:500:12)

agartha-test, 2018-10-27:
I am having the same issue on os Ubuntu 16.04.3 LTS with image raspbian-stretch-lite

thundron, 2018-10-31:
Solved in ac068f353acc53de5ed4edd7b06f5791307abb40

closed 2018-10-31 by thundron