balena-io/etcher #2519: diskpart command failed

martijintro, 2018-10-25:
- **Etcher version: Current** 
- **Operating system and architecture: Win10** 
- **Image flashed: 32GB img** 
- **Do you see any meaningful error information in the DevTools?** 

[image]

cyphermaster, 2018-11-01:
+1
i have the same issue after the windows 10 october update

martijintro, 2018-11-01:
any beta version?

martijintro, 2018-11-19:
no answer, so not interested for an adequate solution so it seems?!?

thundron, 2018-12-12:
@martijintro Sorry for the late reply, this is actually a known issue that goes all the way back to node itself, which has been updated and is included in the latest Electron versions that we're bumping as seen in this PR https://github.com/balena-io/etcher/pull/2578 which is in the works

thundron, 2018-12-12:
@martijintro In the meantime, formatting the disk beforehand *may* help

docholiday, 2018-12-15:
@thundron Just for information - I'm getting the same error in etcher 1.4.8 with my internal Ricoh SD Card Reader. Luckily I could solve it by using an external reader by Kingston...

martijintro, 2018-12-16:
"@martijintro In the meantime, formatting the disk beforehand may help as seen here"

I don't format the disk before using your tool; i just use "HDD Low Level Format Tool" to quick wipe the file system from the drive. (SD card with sandisk USB 3.0 reader)

cyoung, 2019-02-08:
This is still an issue in 1.4.9.

When looking at the disk in "Computer Management", Windows states:

"Offline (The disk is offline because it has a signature collision with another disk that is online)"

After using "Computer Management" to remove the partitions, then disconnecting and re-connecting the USB-to-microUSB adapter, Etcher was able to flash the image on the microSD card.

martijintro, 2019-02-08:
ok thx, will try it as soon as possible...

jimmyeao, 2019-04-02:
Same issue here, deleting the second partition in disk management then connecting to a different USB port solves the issue. Back to win32diskimager until this is fixed..

lurch, 2019-11-19:
> "Offline (The disk is offline because it has a signature collision with another disk that is online)"

Signature collision was also mentioned (with a workaround?) in #2967

thundron, 2019-11-20:
@martijintro Hey there, wondering if this issue is still happening and if you tried with the latest version, since you closed it. We since upgraded Electron, so it might be worth a shot if you'd like to help us figure it out

martijintro, 2019-11-20:
sry. too late, no time for things like that anymore, i'm using another solution now...:-(

nazrhom, 2019-11-20:
@martijintro I will re-open the issue and keep it open if you don't mind. Looks like this is still affecting some users and it makes sense to keep this open until its fully resolved. Let me know if this is ok for you

martijintro, 2019-11-20:
i unsubscribed now, so all ok