balena-io/etcher #2520: Not able to start / use Etcher

alexandervalach, 2018-10-25:
- **Etcher version: 1.4.5** 
- **Linux Mint 19 Cinnamon:** 
[image]

- **I have just started application, could not do anything due to error:** 

TypeError: Expected a number
    at prettyBytes (/opt/Etcher/resources/app.asar/node_modules/pretty-bytes/pretty-bytes.js:18:10)
    at module.exports.getDrivesSubtitle (file:///opt/Etcher/resources/app.asar/generated/gui.js:7477:14)
    at fn (eval at compile (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:15351:15), <anonymous>:4:182)
    at expressionInputWatch (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:16352:31)
    at Scope.$digest (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:18002:34)
    at Scope.$apply (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:18280:24)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:20127:36
    at completeOutstandingRequest (/opt/Etcher/resources/app.asar/node_modules/angular/angular.js:6274:10)
    at /opt/Etcher/resources/app.asar/node_modules/angular/angular.js:6554:7
    at wrapped (/opt/Etcher/resources/app.asar/node_modules/raven-js/dist/raven.js:444:21)

lurch, 2018-10-26:
Duplicate of #2508 ?

thundron, 2018-10-31:
Yes, closing

closed 2018-10-31 by thundron