balena-io/etcher #2521: Multiple mSD cards left in undetectable state as 'no medium' / 'no media' in Linux/Windows

ShiftLeftLogical, 2018-10-25:
- **Etcher version:** 1.4.5
- **Operating system and architecture:** Windows 10 64-bit
- **Image flashed:** 2018-10-09-raspbian-stretch.img
- **Do you see any meaningful error information in the DevTools?** N/A

Ok. I have just caused two different cards to go from fresh out of the package working (formatted to vfat and mounted) to seeing on linux/windows 'no medium' / 'no media' for both mount / disk management tools, respectively. This is Etcher on windows 10.

I figured the first card was just bad so I moved to a different card / manufacturer, same results. Flashing appeared to succeed and so did verification of data but then the card reader / usb media drive is no longer recognized as even having media installed.

Is there any logging mechanism I can submit to help understand what happened and why I cannot see these media devices?

Note that I have used Etcher just fine with this personal card reader/writer on my personal windows machine w/o a single issue, this is not my windows machine so I am not sure if there might be configuration details of the OS which are impacting this (patches/reg/etc.) Are there any registry requirements which are known problems? Are there any BitLocker setting which are known problems?

I see in the event log this following DistributedCOM error which appeared as a popup after the data verification (I believe):

```
The machine-default permission settings do not grant Local Activation permission for the COM Server application with CLSID 
{9BA05972-F6A8-11CF-A442-00A0C90A8F39}
 and APPID 
{9BA05972-F6A8-11CF-A442-00A0C90A8F39}
 to the user XXXX SID (S-1-5-21-3339859558-593924042-3818789186-906134) from address LocalHost (Using LRPC) running in the application container Unavailable SID (Unavailable). This security permission can be modified using the Component Services administrative tool.
```

Windows: When I do a DISKPART LIST DISK I get a disk 1 which has size of 0 and I cannot SELECT.
Linux: If I try to mount /dev/sdb I get the 'no media'

The SD cards were working fine on both OSes. I have tried swapping out the card reader, no luck.

I also tried using SD Formatter Tool but as expected the device doesn't even show up in the drop down list.

lurch, 2018-10-26:
Can you provide more details about the cards (brand, model, size, etc.) and where you bought them from? If even dd on Linux isn't able to fix them (and you've tried different OSes and different card-readers), perhaps you'll be able to return them under warranty?

ShiftLeftLogical, 2018-10-26:
dd was definitely not able to fix them.

I can return both cards and have done so.

The first card was some no-name brand micoSDHC Class 10 16GB, I will check amazon and update the brand name A{XX} something, only recall the first of three characters in the name, I honestly was not surprised with initially finding the card undetectable after flash as it was so cheap but the fact that it happened to the samsung card as well ... made me start to wonder.

The second card was a Samsung microSDHC UHS-I Class 10 32GB. I have another of these cards and will use linux (dd) to flash the image as opposed to Etcher ... if I had more physical cards to spare I might compile and debug the issue but I need one of the cards to actually work. If I get the image on the card via dd once I complete my work. This will be about a week.

lurch, 2018-10-31:
As has been mentioned in other issues, AFAIK Etcher doesn't do anything (in terms of writing to the SD card) that `dd` doesn't; so if `dd` works then Etcher should also work (with the added bonus that Etcher does verification, which dd doesn't).

thundron, 2018-10-31:
In the second case, it looks like it's some permissions handled elsewhere that needs to be addressed, but I can't tell where. Looks like some sort of virtual/limited access container like many companies (and even individuals) use to protect the user environments from external malicious software.
For the first one, I have no clue, it sounds like a faulty sd-card