balena-io/etcher #2524: "Error opening image" for raspberrypi-ua-netinst

milkmiruku, 2018-10-28:
- **Etcher version:** 1.4.5-1
- **Operating system and architecture:** Arch Linux x64
- **Image flashed:** [raspberrypi-ua-netinst-v2.2.1.zip](https://github.com/FooDeas/raspberrypi-ua-netinst/releases)
- **Do you see any meaningful error information in the DevTools?** 

`{"stack":"Error: Invalid archive image\n    at Object.exports.createError (/opt/Etcher/resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/opt/Etcher/resources/app.asar/lib/shared/errors.js:292:18)\n    at hooks.getEntries.then (/opt/Etcher/resources/app.asar/lib/sdk/image-stream/archive.js:181:20)\n    at tryCatcher (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Async._drainQueue (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/opt/Etcher/resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>","message":"Invalid archive image","description":"The archive image should contain one and only one top image file","report":false}
`

Should an archive file only contain one top image file? This would seem, in the context of making it easy for users to get raspberrypi-ua-netinst onto an SD card, too much of a constraint.

luisalcarasr, 2018-10-29:
Hi @milkmiruku 

What would be the utility of recording it on a USB?

milkmiruku, 2018-10-29:
One puts the resultant USB stick into a Raspberry Pi (that has an ethernet connection to an internet gateway), turns the Pi on, the USB wipes the SD card and performs an unattended install of Raspbian plus customisable extras.

thundron, 2018-11-23:
Archives should contain only one image file at least for now.

closed 2018-11-23 by thundron

milkmiruku, 2018-11-23:
@thundron Might there be a rational? Given you note "at least for now.", is it worth opening an issue for a feature request?

thundron, 2018-11-23:
@milkmiruku The main reason I noted "for now" is because *before* going into that we need to recognize the images being written on a deeper level than what we do right now, but we do have plans to extend it that you can see [here](https://github.com/balena-io/etcher/issues/718) and [here](https://github.com/balena-io/etcher/issues/414) (different feature requests but they basically share the same concept of handling the post-write process)