balena-io/etcher #2526: "The disk you inserted was not readable by this computer." after USB Drive is flashed.

GlennTuley, 2018-10-29:
- **Etcher version:** Version 1.4.5 (1.4.5)
- **Operating system and architecture:** macOS Mojave Mac mini (Late 2012) [Macmini6,2], 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3, Intel HD Graphics 4000 1536 MB
- **Image flashed:** linuxmint-19-cinnamon-64bit-v2.iso and several others
- **Do you see any meaningful error information in the DevTools?**

kriekprinsloo, 2018-10-29:
Oh I've seen this one.  It happens to me on a regular basis.  It'll be
fine, try to install from the USB drive anyway.  When I flashed a windows
iso on my USB drive using my MacBook Pro it gave me the same error message,
but installing from the USB drive still works.  If you want to wipe the
drive, use Disk Utility, as the USB drive is still detected by the computer
and it is thus possible to wipe the unreadable partition.

Tell me if this works for you!

SilverWolf32, 2018-10-29:
Don't worry, this is **perfectly normal**. It's not a normal filesystem, so of course it's not a filesystem the computer can read. That's all the message means.

closed 2018-10-31 by thundron