balena-io/etcher #2528: Multilingual interface for GUI and installer

bovirus, 2018-10-30:
- **Etcher version:** 1.4.6 x64
- **Operating system and architecture:** Windows 7 Pro x64 
- **Image flashed:** 

I'd like if I can contribute about the localization for

- Installer
- GUI

Thanks. Best regards.

lurch, 2018-10-31:
Duplicate of #308 ?

thundron, 2018-10-31:
@lurch that's right, use that as a reference for what's to come, closing this.
We're also rewriting the UI with React/[rendition](https://github.com/resin-io-modules/rendition) so this is something that needs some planning beforehand

closed 2018-10-31 by thundron

bovirus, 2018-11-01:
@lurch 
Is not a duplicated request.

I didn't make any request before.
And for now nor in github and and nor in the application there isn't any options for multi-language.

Thanks.

thundron, 2018-11-02:
@bovirus Duplicated as in we already have an issue to keep track of this activity and as you can see, there are many other users like you willing to contribute on localization in the issue @lurch linked.
Don't worry, it's just a matter of time before we start working on it and we'll surely let you know on that issue when it happens :)