balena-io/etcher #2532: Corrupted/Slow SD cards after flash.

Sheepingtin, 2018-11-01:
- **Etcher version:*1.4.6* 
- **Operating system and architecture:*Windows 10* 
- **Image flashed:*OMV_4_Raspberry_Pi_2_3_3Plus.img.xz* 
- **Do you see any meaningful error information in the DevTools?** 
So i have been buying a few micros sd cards for a Raspberry Pi and upon flashing them for the first time with Etcher they all ended up blank and corrupted, Lagging my PC like mad every time i would try and repair them. After several painful hours the Rufus tool managed to make them show up in windows like normal disk again but they are still laggy and impossibly slow (Theres aren't fake cards i'm sure of it) I don't have anything else i can try, Google is dry dose anyone here have a solution or pointers for this?

EinfachAlex, 2018-11-01:
I flashed my SD Card today as well. Now i cant use it anymore. Its not showing up in the Windows Explorer anymore. I cant format it. I tried cleaning via diskpart but this gave me the error that I have no permission to clean the drive.

thejoulethief, 2018-11-07:
@EinfachAlex I had the same issue. I was finally able to format it using this [tool](https://www.sdcard.org/downloads/formatter_4/).

dadapotok, 2018-11-11:
ERR_NAME_NOT_RESOLVED

> @EinfachAlex I had the same issue. I was finally able to format it using this [tool](https://www.sdcard.org/downloads/formatter_4/).

wow, had exactly same issue after trying to use Etcher for xenialpup64-7.5-uefi
SD Memory Card Formatter by Tuxera worked like magic.
thank you @thejoulethief

EinfachAlex, 2018-11-11:
> @EinfachAlex I had the same issue. I was finally able to format it using this [tool](https://www.sdcard.org/downloads/formatter_4/).

That also worked for me, thank you.

thunderpants, 2018-11-23:
I used linux to write an iso image to one of my cards that was screwed up by Etcher in win10.
Write process went through ok and i booted the resulting  linux OS ok
Seems to have brought it back to life by using linux to access the usb card .

thundron, 2018-12-12:
See [this](https://github.com/balena-io/etcher/issues/1426#issuecomment-421938957)

closed 2018-12-12 by thundron