balena-io/etcher #2536: Repository Debian problem with Release file.

jujudusud, 2018-11-04:
Under Debian, Etcher repository seems to be wrong as long as synaptic search the file InRelease in the repository. The file in the repository name is Release.

P.S. Excuse my poor english.

thundron, 2018-11-08:
I think this is due to synaptic itself, I get the same behaviour on totally different packages

jujudusud, 2018-11-11:
Thank for this reply.
apt-get returns the same error...
`Lecture des listes de paquets... Fait
E: Le pilote pour la méthode /usr/lib/apt/methods/https n'a pu être trouvé.
N: Is the package apt-transport-https installed?
E: Impossible de récupérer https://deb.etcher.io/dists/stable/InRelease  
E: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.`
What could I do ?

lurch, 2018-11-12:
`sudo apt-get install apt-transport-https` shoudl fix that problem for you? :smiley:

lurch, 2018-11-12:
But with regards to `Release` vs. `InRelease`, I dug into that a [while ago](https://github.com/balena-io/etcher/issues/2144) and found https://unix.stackexchange.com/questions/387053/debian-9-apt-and-gpg-error-inrelease-the-following-signatures-were-inva
I think I concluded that something about the Bintray setup didn't support Debian9, but I never investigated any further.

jujudusud, 2018-11-12:
> `sudo apt-get install apt-transport-https` should fix that problem for you?

Absolutly ! Resolved for me.

Thanks a lot !

closed 2018-11-12 by jujudusud

Huby1, 2019-06-15:
For my part 'apt-transport-https' did not resolved the issue... 
Impossible to fetch https://deb.etcher.io/dists/stable/etcher/binary-i386/Packages gnutls_handshake() failed: Handshake failed
Any Idea to resolve that please  ?

lurch, 2019-06-15:
See https://github.com/balena-io/etcher/pull/2592#issuecomment-451882927