balena-io/etcher #2540: Debian/Ubuntu repo outdated

pizzadude, 2018-11-09:
Repo only has up to version 1.4.4, but 1.4.6 is available.

Users shouldn't have to manually update.

MeechaelA, 2018-11-12:
Also, the GPG key doesn't work.

thundron, 2018-11-22:
@MeechaelA What do you mean with "doesn't work"? Does it give you any error and if so, could you paste it here?

MeechaelA, 2018-11-24:
Looks like this is no longer a problem, just moved computers so brand new
Debian install. No issues extracting from zip from website. Thanks.

k1nghat, 2018-12-23:
> Side note: we recently changed executable name and for that reason updating manually is the best thing to do _right now_ in order to get the new executable until everything's synced up.
> We'll surely let you know in the forum and have a notice in the readme once it's done, in the meantime I ask you to have a bit more patience.
> I'll keep this issue open as a reference for incoming updates

so do we keep the repo and package we have installed an it will be updated via that or will we have to remove and add a new repo anyways?

thundron, 2018-12-27:
@k1nghat The repo will be the same, but the package name will be different so it's best to uninstall the old & reinstall the new

Mushroomheadbangers, 2019-01-02:
So 1.4.9 dropped recently and the instructions on installing with a package manager (IE: apt) still pulls Etcher 1.4.4. This has been an issue for almost three months.

Seriously...can those steps please be updated and any other necessary changes made to accommodate this? Using Linux Mint 19.1 currently. I'm just going to use the out-dated 1.4.4 until it's resolved.

lurch, 2019-01-02:
https://github.com/balena-io/etcher/issues/2458#issuecomment-450672561 says that the repo will be updated in the next few days :slightly_smiling_face:

LanceHaverkamp, 2019-01-04:
It's not just Ubuntu, Fedora has the same 1.4.4

thundron, 2019-01-04:
We're updating right now (literally) both `deb` and `rpm` packages 👍

thundron, 2019-01-04:
Packages available and README updated in https://github.com/balena-io/etcher/pull/2592. Closing

thundron, 2019-01-04:
Note: new package name is `balena-etcher-electron`

closed 2019-01-04 by thundron

dilbert-tctc, 2019-02-24:
Can you update the Etcher debian repository with version 1.5.0? Please and Thank You.

zero77, 2019-04-02:
It seams to be way out of date again.

zero77, 2019-04-11:
@thundron 
This will help keep the repository up to date using Travis CI.
https://docs.travis-ci.com/user/deployment/bintray/

lurch, 2019-04-11:
@zero77 Etcher _used_ to use TravisCI, but doesn't any more.

zero77, 2019-04-11:
@lurch 
What has Travis CI been replaced with, as there may be an alternative option.

lurch, 2019-04-11:
@zero77 I don't work on Etcher any more, but I _think_ they've replaced it with an internal CI system.

But in any case, in some other issue @thundron said that the Debian / Ubuntu releases are **deliberately** lagging behind the very latest Etcher version, because they want to make sure that a particular version proves to be stable, before pushing it to the repos. (IIRC there's only a 'stable' repo channel, they don't have separate stable and beta channels like they once did. See #1767 and #2076 )