balena-io/etcher #2547: Error "You're not authorized to perform this operation"

Davidw522, 2018-11-19:
Version 1.4.4
WIndows 10x64
when trying to flash a kali linux  to a cd card for a raspberry pi
https://i.imgur.com/L75Ehxo.png
Also tried launching it with admin rights no change.

kriekprinsloo, 2018-11-19:
Huh this could be another npm permissions issue.  Are there any meaningful
messages in the debug console?

thundron, 2018-11-21:
@Davidw522 Did you try opening with admin permissions?
Meanwhile we're investigating this, but Windows needs admin permissions to access drives afaik
See also [here](https://github.com/balena-io/etcher/issues/2548#issuecomment-440489514)

kriekprinsloo, 2018-11-23:
@thundron I don't understand how peter-65 resolved the issue.  It still gave him the permissions error.  Am I missing something here?

IllusionaryX, 2018-12-06:
Same issue here:

Version 1.4.8
WIndows 10x64

Devices attempted to write to:
The Micro SD card: SanDisk Extreme Plus
USB Thumb Drive: Transcend 32GB JetFlash 710

Also ran as Admin. Note... This may be related to another issue that I when searching for an answer...
Etcher is corrupting SD and USB drives. #1426

I only say that because one of the side effects for me after getting this error on a drive that was previously exFAT formatted was that after attempting an attempted write with that error, the previously perfectly usable drive in Windows was rendered unreadable. But I could slap it in my Mac and reformat just fine, rinse and repeat...

Almost feels like it is beginning to write just enough to screw the MBR, then failing (due to some reason) and leaving a crippled disk as a result.

I tried this procedure with both a Micro SD and two USB thumb drive, with the same results. I can say I had earlier versions that worked flawlessly and I have several micro-controllers that are running with cards written from Etcher from this computer. So it would seem either a Windows update, or an Etcher update or both together might have caused an issue. As another note I have written to multiple Transcend JetFlash successfully before though I can not say for certain it was one of these two...

I am a developer of 25 years so I understand that 90% of user complaints are user error, however I do feel there may be some small issue at work here that is causing these errors. But, again I could not say if it is purely a Windows thing (perhaps a "Security Enhancement") or some interaction that has started causing this. But seeing as how you have an intimate knowledge of the code, I figure it is worth documenting for you.

Thanks for the effort I'd love to see an answer for this...

lurch, 2018-12-06:
> Almost feels like it is beginning to write just enough to screw the MBR, then failing (due to some reason) and leaving a crippled disk as a result.

Maybe related to https://www.balena.io/blog/the-perils-of-writing-disk-images-on-windows/  ?

Digicat, 2018-12-25:
I had this issue on Windows 10, and followed the steps here to turn off Windows 10 Controlled Folder Access:  https://www.raspberrypi.org/forums/viewtopic.php?t=186410  which resolved the problem.

closed 2021-08-31 by thundron