balena-io/etcher #2548: corupt SD card or etcher not able to write to card

peter-65, 2018-11-20:
- 1.4.4
- win 10 64 bit pro
- raspbian 2018-11-13 
- No 

problem with formatting or corrupt SD HC card - (also sent to SD format programmers)

Hi (this is being copied to etcher also via GitHub)

I have a 8Gb Samsung micro sd HC card C6 format. 

This card has been used on a raspberry pi (NOBS) formatted card. Last produced on or about 25/01/18 00:12. 

This has 4 partitions when shown on windows 10, 3 have to be formatted by win 10 as they are none windows format. 

The Sd formatter v 5.0.01 will not format to a single partition.  Ready to be used again for the raspberry pie. 

When using the formatter the following is displayed - 

[image]

 

The formatter dose not reformat. The pie (fat32) partition is not over written, this occurs for each of the drives all complete with  

 
[image]

Any ideas or methods to get this card reformatted. 

Etcher gives error when trying to load a new image. 
[image]

 
[image]

thundron, 2018-11-21:
Hi @peter-65 
Did you try giving Etcher admin permissions before running it?

peter-65, 2018-11-21:
Yes etcher works on other cards without problem. I also ran it as an administrator from the program file.by right click "run as admin" to get these errors.

closed 2018-11-21 by peter-65