balena-io/etcher #2555: Etcher Crashes on Launch

steve018, 2018-11-25:
1.4.8
Ubuntu 18.10 64-bit minimal installation
Using .appimage

Whenever I try to open Etcher, I get this error:
`/tmp/.mount_NshRWX/usr/bin/balena-etcher-electron: error while loading shared libraries: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory`

ParthOSDev, 2018-11-25:
Hi,

I suppose you should try sudo apt-get install -y libgconf-2-4 and then try running Etcher again. It solved my problem!

Regards,
?

steve018, 2018-11-25:
That did not fix the issue for me.

thundron, 2018-11-26:
It's strange though, we ship the AppImage with all the needed libraries, including libgconf-2-4.
Did you try installing that @ParthOSDev ?

ParthOSDev, 2018-11-26:
Well it’s not my issue, it is Steve’s and it fixed my problem.

thundron, 2018-11-27:
@steve018 Could you please try installing `libgtk2.0-0:i386` and try running etcher again? It is shipped with the AppImage like the other dependencies, but some users still have issues with them on newer Ubuntu versions

thundron, 2018-11-28:
@mbt45 Did you already try with one/both of the packages suggested above?

thundron, 2018-11-28:
@mbt45 That doesn't sound related to this issue though (this is specific to ubuntu), you can search your error in the other issues and/or open a new one if you can't find it :)

steve018, 2018-11-29:
After installing `libgconf-2-4` and removing it and installing it a second time, the issue is fixed (Didn't work the first time for some reason).

steve018, 2018-11-29:
@ParthOSDev @thundron 
The issue was resolved with a `sudo apt-get install libgconf-2-4 libgtk2.0-common`

closed 2018-11-29 by steve018

lurch, 2018-11-30:
See also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies