balena-io/etcher #2559: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined

jlerigo, 2018-11-26:
- **Etcher version:** 1.4.6
- **Operating system and architecture:** MacOS 10.13.6
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** 

/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 {"stack":"TypeError: Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined\n    at Object.exports.notify (file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:3907:55)\n    at file:///Applications/Etcher.app/Contents/Resources/app.asar/generated/gui.js:3289:27\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at MappingPromiseArray.PromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:125:19)\n    at MappingPromiseArray._filter (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:123:10)\n    at MappingPromiseArray._promiseFulfilled (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/map.js:95:18)\n    at MappingPromiseArray.PromiseArray._iterate (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:113:31)\n    at MappingPromiseArray.init (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise_array.js:77:10)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:574:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:516:17)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Promise._fulfill (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:630:18)\n    at Promise._resolveCallback (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:425:57)\n    at ReductionPromiseArray._resolve (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:51:19)\n    at Promise.completed [as _fulfillmentHandler0] (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/reduce.js:112:15)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:558:21)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:685:18)\n    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>\n    at process._tickCallback (internal/process/next_tick.js:109:7)","message":"Cannot read property 'UPDATE_NOTIFIER_SLEEP_DAYS' of undefined"}

jlerigo, 2018-11-26:
Fixed my upgrading to balendaetcher

closed 2018-11-26 by jlerigo

lurch, 2018-11-30:
Duplicate of #2559